openembedded/meta-openembedded #297: How to deal with npm packages that are only built on host side

hellow554, 2020-12-07:
I want to develop an application using typescript. This app should only be compiled on the hostside, therefore I don't need nodejs on the target.

If I understand the [npm class](http://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/classes/npm.bbclass) correctly i will have nodejs as an runtime dependency. That's why I don't want to use `inherit npm`.

So, what is an good approach to solve my problem? Should I use depend on `nodejs-native` and implement `do_compile` and `do_install` myself? There's a lot of work put in the bbclass, e.g. picking the right target arch, using the correct npm flags as well as having a DEV flag to allow development builds. 

I certanly don't want to implement (copy) them myself. Would it be a good approach to have a seperate `npm-native` class maybe?

What is your opinion about that?

kraj, 2020-12-07:
not sure about npm-native class, perhaps a good idea if we are going to have many host packages to recompile otherwise option 2 is the way to go.

hellow554, 2020-12-08:
So you think it's not worth investing in a `npm-native` class? I certainly don't have enough experience to write such class for the public ;)