openembedded/meta-openembedded #300: rrdtool fails QA on dunfell when configured without perl

kf-chora, 2020-12-14:
Hello there. I hope this is the correct place to report bugs.

If I add `PACKAGECONFIG_remove_pn-rrdtool = "perl"` to my `local.conf` file, then rrdtool will fail QA:
```
ERROR: rrdtool-1.7.2-r0 do_package_qa: QA Issue: rrdtool contains perllocal.pod (/usr/lib/perl5/5.30.1/x86_64-linux/perllocal.pod), should not be installed [perllocalpod]
ERROR: rrdtool-1.7.2-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: Logfile of failure stored in: /home/vagrant/build/tmp/work/core2-64-poky-linux/rrdtool/1.7.2-r0/temp/log.do_package_qa.8708
ERROR: Task (/home/vagrant/poky/meta-openembedded/meta-oe/recipes-extended/rrdtool/rrdtool_1.7.2.bb:do_package_qa) failed with exit code '1'
```

I am not sure why this happens. Same `local.conf` worked for `warrior` branches.

I have created a complete [Vagrantfile.txt](https://github.com/openembedded/meta-openembedded/files/5689513/Vagrantfile.txt) that reproduces the problem. It has been attached. Please remove the `.txt` extension before running vagrant. Github did not allow to upload the file without an extension.

akuster, 2021-01-17:
This patch went in in Zeus.

https://git.openembedded.org/meta-openembedded/commit/meta-oe/recipes-extended/rrdtool?h=zeus&id=5624727547ea8217f5c68f69ff08543a2fc3fa4d

So this issue should affect master.

kf-chora, 2021-01-18:
Agreed. However, it seems that `perllocal.pod` is stillgenerated even when `perl` is removed from the `PACKAGECONFIG` option for the package.