openembedded/meta-openembedded #304: Dunfell build failure mpv-0.32.0-r0 do_configure

akuster, 2020-12-24:
https://errors.yoctoproject.org/Errors/Details/539929/

Seen on all arch builds.  Whats new is building on Ubuntu 20

kraj, 2020-12-24:
seems it has to do with waf and python2 somehow, there are some fixes to address this in master oe-core and meta-oe I believe.

akuster, 2020-12-27:
I did look quick look @ master but nothing jumped out. I run through it again

akuster, 2020-12-27:
Hmm. this meta-oe backport seems to have introduced the failure.

https://git.openembedded.org/meta-openembedded/commit/meta-oe/recipes-multimedia/mplayer?id=5af46f89fcef5c436786ed81978de60f26abe054

kraj, 2020-12-28:
I think there was a corresponding commit in oe-core to waf class.

akuster, 2021-01-16:
ok, found waf.bbclass changes that fix this issue. sending patches core for Dunfell

akuster, 2021-02-07:
in core dunfell
https://git.openembedded.org/openembedded-core/commit/?h=dunfell&id=3ad272ba25c4eba063f372f3bf8c1d3e94e2a966

closed 2021-02-07 by akuster