openembedded/meta-openembedded #310: gatesgarth & Dunfell: qemx86 build failure on luajit

akuster, 2021-01-17:
I have been seeing this for awhile after switching to Ubuntu 20
https://errors.yoctoproject.org/Errors/Details/541244/

/usr/include/linux/errno.h:1:10: fatal error: asm/errno.h: No such file or directory
    1 | #include <asm/errno.h>
      |          ^~~~~~~~~~~~~

kraj, 2021-01-18:
you need 32bit toolchain on build host. see https://github.com/openembedded/meta-openembedded/blob/master/meta-oe/README

akuster, 2021-01-19:
cool. Ubuntu 20.05 needs
sudo apt-get install gcc-multilib

kraj, 2021-01-19:
You can update the readme to include Ubuntu piece of you wish

akuster, 2021-02-07:
patch sent.

akuster, 2021-02-09:
in master. https://git.openembedded.org/meta-openembedded/commit/?id=973fe410d238e0c361f8bc4d9ba7915464217e22

closed 2021-02-09 by akuster