openembedded/meta-openembedded #311: wx-widgets sdk build not working

jascase901, 2021-01-25:
I can build applications that depend of wx using bitbake. But if I try to make an sdk. and build a sample application get 

```
CMake Error at /opt/tegrademo-mender/3.1+snapshot/sysroots/x86_64-tdsdk-linux/usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
  wxWidgets_INCLUDE_DIRS)
Call Stack (most recent call first):
  /opt/tegrademo-mender/3.1+snapshot/sysroots/x86_64-tdsdk-linux/usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
  /opt/tegrademo-mender/3.1+snapshot/sysroots/x86_64-tdsdk-linux/usr/share/cmake-3.18/Modules/FindwxWidgets.cmake:1008 (find_package_handle_standard_args)
  CMakeLists.txt:13 (find_package)
```

Looking at my sys root directory. there is no wx-config, which is hwat findwxwidgets uses to set the include, and library dirs. There is a wxrc, which is functionally the same thing as wx-config, but trying to use it returns my system wx libraries in /usr/local

gcdcintra, 2022-04-15:
I had a similar problem. Solved that by adding `DEPENDS += "wxwidgets"` to my recipe .bb file.

With that, I could compile the recipe using standard cmake syntax described at wxwidgets docs:
```
find_package(wxWidgets COMPONENTS core base REQUIRED)
include(${wxWidgets_USE_FILE})
...
target_link_libraries(APP PUBLIC ${wxWidgets_LIBRARIES})
```

kraj, 2023-01-13:
you have to generate the SDK after doing these changes.