openembedded/meta-openembedded #327: UFW in dunfell: 'python': No such file or directory

jostor, 2021-03-26:
When running UFW 0.33, built from the dunfell branch, I get this error:` /usr/bin/env: 'python': No such file or directory.` I guess it is because python2 is removed from the dunfell branch. But shouldnt it be possible to use UFW with python3? And shouldn't that be default on the dunfell branch?

shr-project, 2021-03-26:
Using python3 with dunfell would be definitely preferred. Patches are always welcome.

akuster, 2021-03-31:
This should fail on master as well. There have not been any changes to the recipe zeus

Silcet, 2021-04-23:
I actually have a patch for this. Let me clean it up and I'll link the PR here

Silcet, 2021-04-23:
I have fixed this issue for my system. I'll create a pull request soon