openembedded/meta-openembedded #357: Build opencv fault

sss22213, 2021-05-13:
hi here,

I got some error while build opencv.

Build Command:
```bash=
bitbake opencv
```

Error message:
```bash=
mp/work/cortexa72-poky-linux/opencv/4.5.2-r0/recipe-sysroot/usr/include/eigen3/Eigen/src/Core/products/TriangularMatrixVector.h:105:24: note: by argument 5 of type 'const double*' to 'static void Eigen::internal::triangular_matrix_vector_product<Index, Mode, LhsScalar, ConjLhs, RhsScalar, ConjRhs, 1, Version>::run(Index, Index, const LhsScalar*, Index, const RhsScalar*, Index, Eigen::internal::triangular_matrix_vector_product<Index, Mode, LhsScalar, ConjLhs, RhsScalar, ConjRhs, 1, Version>::ResScalar*, Index, const ResScalar&) [with Index = long int; int Mode = 6; LhsScalar = double; bool ConjLhs = false; RhsScalar = double; bool ConjRhs = false; int Version = 0]' declared here
|   105 | EIGEN_DONT_INLINE void triangular_matrix_vector_product<Index,Mode,LhsScalar,ConjLhs,RhsScalar,ConjRhs,RowMajor,Version>
|       |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

```bash=
Build Configuration:
BB_VERSION           = "1.51.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi4-64"
DISTRO               = "poky"
DISTRO_VERSION       = "3.3+snapshot-bb490c7ea9170fc1b3c2dd15466daf6a9fad735c"
TUNE_FEATURES        = "aarch64 armv8a crc crypto cortexa72"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "master:bb490c7ea9170fc1b3c2dd15466daf6a9fad735c"
meta-raspberrypi     = "master:d4d1441bd217f1a3dc353a6c20eadceb819c8e37"
meta-oe              
meta-multimedia      
meta-networking      
meta-python          = "master:3c259f977e8647ac9c54097d0faab70719a5553e"
meta-qt5             = "master:a2882fecc96019781880c38fc6a81a9b71c944ad"
```

Thanks in advance.

kraj, 2021-05-13:
```
aarch64-poky-linux-g++: fatal error: Killed signal terminated program cc1plus
```

this tells me that your system ran out of memory. So lower your parallelism or add more RAM to your build system.

sss22213, 2021-05-13:
hi kraj,

Thank you for the solution.

I solve the problem.

closed 2021-05-13 by sss22213