openembedded/meta-openembedded #372: How to put libgpiod v1.6(hardknott yocto release) in zeus release

neuberfran, 2021-06-12:
Hi,

I have a great knowledge of this tutorial: https://hub.mender.io/t/technexion-pico-pi-imx7/136
As can be seen he goes to Zeus (libgpiod version = 1.4)

I need libgpiod v1.6 in my project to configure pull-up/pull-down

I did the copy procedures, etc... and got the issue ERROR: Layer openembedded-layer is not compatible with the core layer which only supports these series: zeus (layer is compatible with hardknott)

How to solve (psl)

[image]

:

kraj, 2021-06-12:
Either upgrade your distro to hardknott release or backport the needed recipes to Zeus branch mixing layers across releases is a big no no unless allowed by layer compatible options explicitly

neuberfran, 2021-06-12:
@kraj  You said: "backport the needed recipes to Zeus branch". 

The creator of libgpiod also suggested this to me. I'm a beginner and I don't know how to do this the best way. Could you (please) give me the step by step for me to do this?

e.g.: The creator of libgpiod said he is too busy and can't do it.

kraj, 2021-06-12:
start by replacing libgpiod direcrory in zeus with one from master and see how it goes.

neuberfran, 2021-06-12:
@kraj 

ERROR: ParseError at /home/neuberfran/mender-nxp3/build/../sources/meta-openembedded/meta-oe/recipes-support/remmina/remmina_1.3.6.bb:18: Could not inherit file classes/mime-xdg.bbclass

kraj, 2021-06-12:
did you copy whole meta-openembedded dir ? I suggested only libdpiod recipes

neuberfran, 2021-06-12:
@kraj I had actually done this in the first text of this issue (-b zeus). This time I just copied the libgpiod directory which contains 1.6 files (including the files folder)

neuberfran, 2021-06-12:
@kraj
The new meta-openembedded from yocto 3.3 have
remmina_1.4.17.bb file in remmina folder. The zeus have remmina_1.3.6.bb. I change but not Solved. Same issue. 
The file
mime-xdg.bbclass doesn't even show up in the mender-nxp project (which works, but with libgpiod v1.4). The mime-xdg.bbclass file only appears in technexion's yocto project for imx7-pico (without the d) and which also only has libgpiodv1.4. You can see in the photo below, result of the find command

https://github.com/TechNexion/tn-imx-yocto-manifest/tree/zeus_5.4.y-next

[image]

neuberfran, 2021-06-13:
@kraj 

Below are the latest errors I've been getting:
(there is also 1 link with libgpiod files for me to manually put on my device)

ERROR: Nothing RPROVIDES 'python3-packaging' (but /home/neuberfran/mender-nxp4/build/../sources/meta-openembedded/meta-oe/recipes-support/libgpiod/libgpiod_1.6.3.bb RDEPENDS on or otherwise requires it)

ERROR: Required build target 'core-image-base' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-base', 'libgpiod', 'python3-packaging']

http://ftp.iij.ad.jp/pub/linux/kernel/software/libs/libgpiod/

closed 2025-06-25 by neuberfran