openembedded/meta-openembedded #377: Branch name to main in recipies

kallmanjaa, 2021-06-19:
Some of the recipes source uri is changed from branch name=master to branch name=main. Please could you guys fix this in all branches. Because of this our ci build is failing for out yocto project.

shr-project, 2021-06-19:
You should at least say which recipes are failing for you and your ci. All builds fine for me, but I have well populated premirror, so issues like this aren't detected by ci builds I'm running.

kallmanjaa, 2021-06-20:
It is failing in the recipie ostree in recipies-extended in Zeus branch.
Actually we dint mirror the sources , but next time we would definitely do it.

shr-project, 2021-06-20:
zeus is already EOL, see https://wiki.yoctoproject.org/wiki/Releases , you can easily fix ostree SRC_URI with local .bbappend and start planing to upgrade to newer release.

closed 2021-06-20 by shr-project