openembedded/meta-openembedded #407: graphviz is using host includes/libs [hardknott], when tcl is installed on the host

kranzo, 2021-08-07:
if build in crops(ubuntu-20.04) everything works properly

on my baremetal workstation (ubuntu-20.04) tlc8.6 is installed, resulting into QA-Error because host paths are included.

removing this patch leads to a error free compilation: https://github.com/openembedded/meta-openembedded/blob/9b23e1f76b76a43ea87e6364c23f29a21da81e61/meta-oe/recipes-graphics/graphviz/graphviz/0001-Set-use_tcl-to-be-empty-string-if-tcl-is-disabled.patch

so my question is why would i configure tcl if tcl is explicitly disabled? Because setting the string to "" is actually the default (use_tcl) config. I feel like the patch is not doing  what is said in the commit-msg

michalfita, 2023-01-10:
On Ubuntu 22.04 with Yocto Honister (brought in by PetaLinux v2022.1) `graphviz` was failing for me with:
```
/usr/lib//tcl8.6/tclConfig.sh: line 2: dpkg-architecture: command not found
```
Removing this patch as above issue mentions solves the problem for me.

The recipe in use is `components/yocto/layers/meta-openembedded/meta-oe/recipes-graphics/graphviz/graphviz_2.44.1.bb`.

Thanks, @kranzo.

kraj, 2023-01-10:
@kranzo @michalfita thanks for report. `honister` is EOL but if you can reproduce it with `master` or `kirkstone`, that would be awesome. I am using archlinux host and has `tclsh8.6` installed but I do not see this error on master branch

Youras, 2024-04-11:
Somehow removing this patch, didn't solve the error. But here is a dirty fix:

The cause of the error is sourcing a tcl config script located in `/usr/lib/tcl8.6/tclConfig.sh` as the error prompt says
```
#! /bin/sh
. /usr/lib/`dpkg-architecture -qDEB_HOST_MULTIARCH`/tcl8.6/tclConfig.sh
```

Now `dpkg-architecture -qDEB_HOST_MULTIARCH` can be resolved by simply running this part in a console and replacing this command by the prompt of the console.
In my case the prompt is `x86_64-linux-gnu`. Thus the final file may look like this:
```
#! /bin/sh
. /usr/lib/`dpkg-architecture -qDEB_HOST_MULTIARCH`/tcl8.6/tclConfig.sh
```
Now building should work fluently :)