openembedded/meta-openembedded #408: fbida do_compile step fails (missing dependencies)

brampetersen, 2021-08-09:
Hi,

I'm currently upgrading from zeus to hardknott, and fbida 2.14 (was 2.10) fails to compile due to missing dependencies. See log below:

```
ERROR: fbida-2.14-r0 do_compile: oe_runmake failed
ERROR: fbida-2.14-r0 do_compile: Execution of '/home/developer/build/tmp-glibc/work/cortexa7t2hf-neon-mx7ulp-fslc-linux-gnueabi/fbida/2.14-r0/temp/run.do_compile.28541' failed with exit code 1:
checking for X11 app-defaults prefix ... /etc/X11
checking for linux/fb.h ... yes
checking for cairo-gl (using pkg-config) ... no
checking for pcd_open in pcd ... no
checking for DGifOpenFileName in gif ... yes
checking for libjpeg version ... 62
checking for fopencookie ... yes
checking for sane_init in sane ... no
checking for curl_easy_init in curl ... yes
checking for lirc_init in lirc_client ... no

Make.config written, edit if needed

Build dependencies missing for fbi and/or fbpdf.
  fbi   needs:  freetype2 fontconfig libdrm libexif libpng libtiff-4 pixman-1
  fbpdf needs:  libdrm poppler-glib gbm egl epoxy pixman-1
Please install.  Try 'make yum', 'make dnf' or 'make apt-get' (needs sudo).
make: *** [/home/developer/build/tmp-glibc/work/cortexa7t2hf-neon-mx7ulp-fslc-linux-gnueabi/fbida/2.14-r0/fbida-2.14/GNUmakefile:32: deps] Error 1
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/developer/build/tmp-glibc/work/cortexa7t2hf-neon-mx7ulp-fslc-linux-gnueabi/fbida/2.14-r0/temp/log.do_compile.28541
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 8 STRIP= srcdir=/home/developer/build/tmp-glibc/work/cortexa7t2hf-neon-mx7ulp-fslc-linux-gnueabi/fbida/2.14-r0/fbida-2.14 -f /home/developer/build/tmp-glibc/work/cortexa7t2hf-neon-mx7ulp-fslc-linux-gnueabi/fbida/2.14-r0/fbida-2.14/GNUmakefile HAVE_CAIRO_GL=no
| checking for X11 app-defaults prefix ... /etc/X11
| checking for linux/fb.h ... yes
| checking for cairo-gl (using pkg-config) ... no
| checking for pcd_open in pcd ... no
| checking for DGifOpenFileName in gif ... yes
| checking for libjpeg version ... 62
| checking for fopencookie ... yes
| checking for sane_init in sane ... no
| checking for curl_easy_init in curl ... yes
| checking for lirc_init in lirc_client ... no
|
| Make.config written, edit if needed
|
| Build dependencies missing for fbi and/or fbpdf.
|   fbi   needs:  freetype2 fontconfig libdrm libexif libpng libtiff-4 pixman-1
|   fbpdf needs:  libdrm poppler-glib gbm egl epoxy pixman-1
| Please install.  Try 'make yum', 'make dnf' or 'make apt-get' (needs sudo).
| make: *** [/home/developer/build/tmp-glibc/work/cortexa7t2hf-neon-mx7ulp-fslc-linux-gnueabi/fbida/2.14-r0/fbida-2.14/GNUmakefile:32: deps] Error 1
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/home/developer/build/tmp-glibc/work/cortexa7t2hf-neon-mx7ulp-fslc-linux-gnueabi/fbida/2.14-r0/temp/run.do_compile.28541' failed with exit code 1:
| checking for X11 app-defaults prefix ... /etc/X11
| checking for linux/fb.h ... yes
| checking for cairo-gl (using pkg-config) ... no
| checking for pcd_open in pcd ... no
| checking for DGifOpenFileName in gif ... yes
| checking for libjpeg version ... 62
| checking for fopencookie ... yes
| checking for sane_init in sane ... no
| checking for curl_easy_init in curl ... yes
| checking for lirc_init in lirc_client ... no
|
| Make.config written, edit if needed
|
| Build dependencies missing for fbi and/or fbpdf.
|   fbi   needs:  freetype2 fontconfig libdrm libexif libpng libtiff-4 pixman-1
|   fbpdf needs:  libdrm poppler-glib gbm egl epoxy pixman-1
| Please install.  Try 'make yum', 'make dnf' or 'make apt-get' (needs sudo).
| make: *** [/home/developer/build/tmp-glibc/work/cortexa7t2hf-neon-mx7ulp-fslc-linux-gnueabi/fbida/2.14-r0/fbida-2.14/GNUmakefile:32: deps] Error 1
| WARNING: exit code 1 from a shell command.
|
ERROR: Task (/home/developer/sources/meta-openembedded/meta-oe/recipes-graphics/fbida/fbida_2.14.bb:do_compile) failed with exit code '1'
```

It seems that not all of the DEPENDS are added to the `fbida_2.14.bb` file but I can't find which one. I tried adding some in a .bbappend file but with no luck.

Thanks in advance!

simoneScaravati, 2025-09-22:
I wouldn't suggest using this patch on the master branch, unless the scope is to remove the fpdf tool from the recipe; in that case, it should be okay. 
Anyway, I just tested the **fbi** tool compiled with my patch under Kirkstone, using a custom image that uses Qt6, and **it works**. 

For completion, this is the fbida_%.bbappend file: 
```
FILESEXTRAPATHS:prepend := "${THISDIR}/files:"

SRC_URI += "file://0010-GNUmakefile-remove-fbpdf.patch"

DEPENDS:append:pn-fbida = " freetype fontconfig libexif libpng tiff pixman libpng tiff"

REQUIRED_DISTRO_FEATURES = "${@bb.utils.contains('PACKAGECONFIG', 'pdf', 'opengl', '', d)}"
```

If you agree to remove the fbpdf tool from the recipe, I'll create a PR. 
Cheers

simoneScaravati, 2025-09-22:
EDIT: I found out that in the scarthgap branch, the issue is already solved correctly, making the fbpdf tool really optional: 
https://github.com/openembedded/meta-openembedded/blob/scarthgap/meta-oe/recipes-graphics/fbida/files/0001-meson.build-make-fbpdf-build-optional.patch
But it is a meson.build file, not GNUmakefile. So for the Kirkstone branch, I think a backport is needed.  
If you want, I could try to take care of it.

kraj, 2025-09-22:
> EDIT: I found out that in the scarthgap branch, the issue is already solved correctly, making the fbpdf tool really optional: [...]

Please do so.