openembedded/meta-openembedded #413: gpsd: 1024 week backward time jump due October 2021

mikini, 2021-08-16:
Is work being done on bringing a fix into gpsd of meta-oe to the [recently discovered issue in gpsd >=3.20, gpsd <= 3.22](https://gitlab.com/gpsd/gpsd/-/issues/144) (introduced on [2019-10-08](https://gitlab.com/gpsd/gpsd/-/issues/144#note_633544667)) causing it to unconditionally relay a 1024 week backwards time jump on 2021-10-24 (destination year 2002)?

The bug is triggered by [leap seconds not having been inserted](https://datacenter.iers.org/data/latestVersion/16_BULLETIN_C16.txt) for an unusual long (>2180 weeks) period (really: earth spinning faster, [maintainer points to global warming as the cause](https://gitlab.com/gpsd/gpsd/-/issues/144#note_633612324)).

A [fix has been comitted](https://gitlab.com/gpsd/gpsd/-/commit/7f30d88d04dc62b8bd6265ad1d09d72d220f97f6) and was [released in gpsd v3.23 on 2021-08-06](https://lists.nongnu.org/archive/html/gpsd-announce/2021-08/msg00000.html). Official backports to older releases [will not be made available](https://gitlab.com/gpsd/gpsd/-/issues/144#note_642351426).

Looking at versioning of the gpsd recipe, v3.20 is in [master](https://layers.openembedded.org/layerindex/recipe/1120/), and the two latest release branches of Yocto (ie. [gatesgarth](https://layers.openembedded.org/layerindex/recipe/153335/) and [hardknott](https://layers.openembedded.org/layerindex/recipe/169887/)) so those will expose the bug.

# Mentions
* [LWN](https://lwn.net/Articles/865044/)
* [HN](https://news.ycombinator.com/item?id=28041062)
* [debian backport](https://salsa.debian.org/debian-gps-team/pkg-gpsd/-/commit/2df40c7640dcbc5cbc48969bec44932623ef243b) ([fix](https://salsa.debian.org/debian-gps-team/pkg-gpsd/-/commit/c6af361ae1ba59dc8d3ac8783faabcd5a059c092))
* [USN-5035](https://ubuntu.com/security/notices/USN-5035-1)
  * [launchpad #1938730](https://bugs.launchpad.net/ubuntu/+source/gpsd/+bug/1938730)

kraj, 2021-08-16:
@mikini thanks for report. We still maintain dunfell and hardknott releases, so this would need fixing there @akuster might have better ideas as to what can be done.

akuster, 2021-08-16:
@Khem, this also affects Master unless we update to the later version 3.23

kraj, 2021-08-16:
> @Khem, this also affects Master unless we update to the later version 3.23

right ok. Then perhaps we need to update this recipe on master first.

mtudan, 2021-08-19:
Hey guys, I just prepared a pull request on this matter:
https://github.com/openembedded/meta-openembedded/pull/415