openembedded/meta-openembedded #426: how to build a light gnome desktop for my imx8 platform

jwzl, 2021-08-30:
hello,
     I want to build light gnome desktop for my imx8 platform, and it 's based on weston. and how can  I do for this ? thank you very much!

Best Regards
Qing

kraj, 2021-09-01:
I think there is no image like core-image-minimal-xfce which is readily available. But I am sure you can cook one up using components from meta-gnome where you can add
`IMAGE_INSTALL += "packagegroup-gnome-desktop"` to
`core-image-x11`

jwzl, 2021-09-02:
I see the packagegroup-gnome-desktop based on x11, but my platform is based wayland
it's okay ? thanks
@kraj

jwzl, 2021-09-02:
@kraj I can't find the packagroup( packagegroup-gnome-desktop) in this layer

jwzl, 2021-09-02:
packagegroup-xfce-base is also not found!

schnitzeltony, 2021-09-02:
On Mon, Aug 30, 2021 at 3:46 AM CH_Qing ***@***.***> wrote:
>
> hello,
> I want to build light gnome desktop for my imx8 platform, and it 's based on weston. and how can I do for this ? thank you very much!
>
> Best Regards
> Qing

As the maintainer of meta-gnome:
For me all variants of gnome desktop are broken for a while now. I did
not invest much efforts in the past because gjs and dependents mutter
and and gnome-session were stuck to old version due to impossible
upgrade of mozjs. Mozjs could not be upgraded due to missing rust
support in oe-core.
The rust situation has changed few days ago: Thanks the excellent work
of oe-core/meta-rust people we now have rust support in oe-core.
I have started to work on mozjs-78 but am still far from something working.
So long talk: Even if you manage to pick up the necessary bits for
gnome on wayland don't expect it to work currently. Hope this changes
soon but be aware that my work on gnome is done in my free time - so
it might last.

Cheers

Andreas

crab2313, 2021-10-30:
@kraj 

What's the reason to upgrade mozjs from 68 to 91? It seems that gjs is depending on mozjs78. Should we use independent package names such as mozjs78 and mozjs91?

schnitzeltony, 2021-10-30:
If I remember correctly @kanavin - the author of mozjs-91 - said, the reasons were:

* latest polkit requires 91
* 78 did not build with python 3.10

Hope to come up with fresh gnome soon [1]. Have to rework that further to build cleanly with all (corner) arches.

[1] https://github.com/schnitzeltony/meta-openembedded/tree/de-rotten-meta-gnome-part-I