openembedded/meta-openembedded #499: syntax error in meta-python/recipes-devtools/python/python3-fasteners_0.16.3.bb

pchand20, 2021-12-06:
seems there is some syntax error in meta-python/recipes-devtools/python/python3-fasteners_0.16.3.bb recipe 

here is syntax which is giving some error.
**RDEPENDS:${PN} += "**\
    ${PYTHON_PN}-logging \
    ${PYTHON_PN}-fcntl \
"
correct syntax working for me is following. also aligned with yocto manual. 

**RDEPENDS_${PN} += "\**
    ${PYTHON_PN}-logging \
    ${PYTHON_PN}-fcntl \
"

shr-project, 2021-12-06:
See https://github.com/openembedded/meta-openembedded/pull/497 and update your bitbake revision.

pchand20, 2021-12-08:
seems some latest version of bitbake needed for compatibility to parse new syntax. could not find exactly which version of bitbake have this support but some oct-9 commit on 1.4 works for me. closing it.

closed 2021-12-08 by pchand20

shr-project, 2021-12-08:
https://github.com/ros/meta-ros/pull/902 linked from the other PR lists the exact commits where the compatibility was introduced.