openembedded/meta-openembedded #529: VLC dunfell, unable to play hd video

adityasingh1993, 2022-02-07:
I enabled the hardware acceleration while building the vlc for Raspberry pi 3b+, but still HD videos are getting stucked.

adityasingh1993, 2022-02-07:
@kraj I buid an image for Raspberry pi 3b+ using yocto project. While building it I have used VLC instead of OMXplayer for playing videos, and enabled the hardware acceleration using `--enable-omxil --enable-omxil-vout --enable-rpi-omxil --enable-mmal` But the problem is with the 1080x1920 resolution videos. VLC is not able to play these videos. I used --codec omxil --vout omxil_vout while playing the video, but it still showing avcodec as decoder and video is getting stucked. So basically I am getting `avcodec decoder error: more than 5 seconds of late video`

kraj, 2022-02-07:
I understand that, however, I am wondering if we have set of patches that we can apply on top of the core provided recipe.

adityasingh1993, 2022-02-07:
@kraj kindly help me if you have any patch or link related to this.

adityasingh1993, 2022-02-09:
@kraj any solution or patch for this?

kraj, 2022-02-09:
I think looking at the raspbian fork of vlc might give some hints

closed 2022-04-16 by halstead