openembedded/meta-openembedded #537: Updaste matplotlib 1.1.0 to 2.0.2 in morty

klasyc, 2022-02-25:
Hello guys,

I am using (quite aging) `morty` branch of this repository. There is `matplotlib` version 1.1.0 which does not compile correctly. To workaround this problem I cherry-pick commit 10c8c39c25d902ab5268fa48ac2bc0e42e65b179 every time I checkout the repository. Would you consider to include this commit in `morty`?

Thank you in advance!
Klasyc

kraj, 2022-02-25:
morty is dead, I will leave this upto @akuster to decide if he is ok with it

shr-project, 2022-02-25:
dead and major version upgrades are usually rejected even for stable branches which are still activelly maintained. Why don't you put matplotlib_2.0.2.bb in some meta-backport-from-2017 layer if you cannot easily upgrade from morty release to something still supported?

klasyc, 2022-02-25:
> dead and major version upgrades are usually rejected even for stable branches which are still activelly maintained. Why don't you put matplotlib_2.0.2.bb in some meta-backport-from-2017 layer if you cannot easily upgrade from morty release to something still supported?

I was considering to do that, but first I wanted to try share my modifications with other developers. I will add this modification to my custom layer as you propose.

Thank you.