openembedded/meta-openembedded #539: salsa.debian.org is down impacting yocto builds

geissonator, 2022-03-01:
It's been down for a few days now (https://micronews.debian.org/2022/1645992519.html). Time to look for an alternative?

```
-c gc.autoDetach=false fetch -f --progress https://salsa.debian.org/debian/ca-certificates.git refs/*:refs/* failed with exit code 128, no output
ERROR: ca-certificates-20211016-r0 do_fetch: Fetcher failure for URL: 'git://salsa.debian.org/debian/ca-certificates.git;protocol=https;branch=master'. Unable to fetch URL from any source.
```

akuster, 2022-03-01:
is this the same?
https://github.com/openSUSE/ca-certificates

geissonator, 2022-03-01:
I'm not sure, a grep shows a few git repos from it:
```
meta-filesystems/recipes-utils/fatresize/fatresize_1.1.0.bb:SRC_URI = "git://salsa.debian.org/parted-team/fatresize.git;protocol=https;branch=master \

meta-networking/recipes-support/ifenslave/ifenslave_2.13.bb:SRC_URI = "git://salsa.debian.org/debian/ifenslave.git;protocol=https;branch=main"

meta-oe/recipes-devtools/python/python3-apt_2.3.0.bb:SRC_URI = "git://salsa.debian.org/apt-team/python-apt.git;protocol=https;branch=main"

meta-oe/recipes-support/consolation/consolation_0.0.8.bb:SRC_URI = "git://salsa.debian.org/consolation-team/consolation.git;branch=master"
```

kraj, 2022-03-01:
could we see if we can use `https://code.launchpad.net/` in interim ?

geissonator, 2022-03-02:
Git repos are back up so a moot point now I guess.

closed 2022-03-02 by geissonator