openembedded/meta-openembedded #546: error compile of mpd after update to 23 ?!!

fairbird, 2022-03-10:
Need to update libupnp to 1.8 ...
> | Dependency libsystemd skipped: feature systemd disabled
| Dependency libupnp found: NO found 1.6.25 but need: '>= 1.8'
| Run-time dependency libupnp found: NO
| 
| ../git/src/lib/upnp/meson.build:20:2: ERROR: Invalid version of dependency, need 'libupnp' ['>= 1.8'] found '1.6.25'.

kraj, 2022-03-10:
I can build fine on master branch, which release are you building ?

fairbird, 2022-03-10:
master also
https://github.com/fairbird/openpli-dreambox-oe-core/commit/b27a23db8fafd748eafae4c8ec4d463106fcfd70#diff-57bdbdb058e84d69756cf087007d05f6c47ca34ee806ca6221de470449d4718a

kraj, 2022-03-11:
interesting, I do not see this issue here on several machines from arm, riscv, x86, mips, so perhaps more info is needed.

fairbird, 2022-03-11:
I have clean build directory completely and I have same issue ...
I try to compile it for arm V7 for dreambox device and also Mipsel ..Same error message

kraj, 2022-03-11:
Do you have other layers providing older libupnp ? I am afraid I can not help you unless there is some
More information here

fairbird, 2022-03-11:
No there is no extra providing for old libupnp in my source
https://github.com/fairbird/openpli-dreambox-oe-core/blob/python3/meta-openpli/conf/distro/openpli.conf
https://github.com/fairbird/openpli-dreambox-oe-core/blob/python3/meta-openpli/conf/layer.conf

fairbird, 2022-03-11:
Oops ... I think this is the reason of the issue old version I have forgotten to delete it
https://github.com/fairbird/openpli-dreambox-oe-core/tree/python3/meta-openpli/recipes-connectivity/libupnp

Thank you for your time ...

closed 2022-03-12 by fairbird