openembedded/meta-openembedded #590: pcp useradd seems to be broken on kirkstone

b1czu, 2022-08-08:
Hi,
seems to be that the `USERADD_PARAM:${PN}` for _pcp_ package is failing for some reason. The **pcp** user is not added to the target system and pcp services fail to start. If I additionally add _pcp-testsuite_ package to my image then **pcpqa** user is added to the target system without any problem but **pcp** user is still missing.
If I manually change pcp recipe from:
```
USERADD_PACKAGES = "${PN}"
USERADD_PARAM:${PN} = "--system --home ${localstatedir}/lib/pcp --no-create-home \
                       --user-group pcp"
```
to:
```
USERADD_PACKAGES = "pcp"
USERADD_PARAM:pcp = "--system --home ${localstatedir}/lib/pcp --no-create-home \
                       --user-group pcp"
```
then **pcp** user is added correctly.

I'm building using KAS tool for x86-64 target and using latest meta-oe from kirkstone branch so I'm using this version of pcp recipe: https://github.com/openembedded/meta-openembedded/blob/654af6d6ecf49c6641de090950043563071aedda/meta-oe/recipes-support/pcp/pcp_5.3.6.bb.

b1czu, 2022-08-08:
@kraj @leimaohui

kraj, 2022-08-08:
@b1czu can you try following patch and report back

```patch
diff --git a/meta-oe/recipes-support/pcp/pcp_5.3.6.bb b/meta-oe/recipes-support/pcp/pcp_5.3.6.bb
index 48d536e1d6..0543d77a78 100644
--- a/meta-oe/recipes-support/pcp/pcp_5.3.6.bb
+++ b/meta-oe/recipes-support/pcp/pcp_5.3.6.bb
@@ -34,7 +34,7 @@ USERADD_PACKAGES = "${PN}"
 USERADD_PARAM:${PN} = "--system --home ${localstatedir}/lib/pcp --no-create-home \
                        --user-group pcp"

-USERADD_PACKAGES = "${PN}-testsuite"
+USERADD_PACKAGES += "${PN}-testsuite"
 USERADD_PARAM:${PN}-testsuite = "--system --home ${localstatedir}/lib/pcp/testsuite --no-create-home \
                        --user-group pcpqa"
```

b1czu, 2022-08-09:
> @b1czu can you try following patch and report back
> 
> ```diff
> diff --git a/meta-oe/recipes-support/pcp/pcp_5.3.6.bb b/meta-oe/recipes-support/pcp/pcp_5.3.6.bb
> index 48d536e1d6..0543d77a78 100644
> --- a/meta-oe/recipes-support/pcp/pcp_5.3.6.bb
> +++ b/meta-oe/recipes-support/pcp/pcp_5.3.6.bb
> @@ -34,7 +34,7 @@ USERADD_PACKAGES = "${PN}"
>  USERADD_PARAM:${PN} = "--system --home ${localstatedir}/lib/pcp --no-create-home \
>                         --user-group pcp"
> 
> -USERADD_PACKAGES = "${PN}-testsuite"
> +USERADD_PACKAGES += "${PN}-testsuite"
>  USERADD_PARAM:${PN}-testsuite = "--system --home ${localstatedir}/lib/pcp/testsuite --no-create-home \
>                         --user-group pcpqa"
> ```

I can't believe I didn't notice it! I can confirm that with this patch user `pcp` is added correctly. However, some pcp services still do not start correctly (at least with systemd enabled target system). 
```
systemctl --failed
● pmcd.service          loaded failed failed Performance Metrics Collector Daemon
● pmie.service          loaded failed failed Performance Metrics Inference Engine
● pmie_farm.service     loaded failed failed pmie farm service
● pmlogger.service      loaded failed failed Performance Metrics Archive Logger
● pmlogger_farm.service loaded failed failed pmlogger farm service
```
There seems to be some problem with writing pcp's pid files to `/var/run/` (pid file is created but not filled with process id) and systemd is restarting services endlessly, but probably this should go to another issue report.