openembedded/meta-openembedded #606: Kirkstone branch build is currently broken when trying to compile any npm recipe

aurelihein, 2022-09-13:
If you try to compile a recipe that use npm on yocto _poky_ **kirkstone** and _meta-openembedded_ **kirkstone** branch the build will fail with :
`ERROR: Nothing PROVIDES 'nodejs-oe-cache-native' (but /srv/yocto/poky/**** DEPENDS on or otherwise requires it). Close matches:", "  nodejs-native", "ERROR: Required build target '****' has no buildable providers."`
It seems introduced by the last commit on yocto _poky_ **kirkstone** : [HERE](https://git.yoctoproject.org/poky/commit/?h=kirkstone&id=b53aff215effa6fdb02d3ac90d6ea654bd43e0eb)
It works though if you switch from **kirkstone** to **master** meta-openembedded
Am I sharing this in the right place ?
Thanks for your feedback
Aurelien BOUIN

shr-project, 2022-09-13:
@akuster meta-oe-cache-native cherry-picked to meta-oe/kirkstone sent https://lists.openembedded.org/g/openembedded-devel/message/98787

aurelihein, 2022-09-13:
Thank you @shr-project I guess we just need to wait now :+1:

mvargasevans, 2023-02-01:
Hi
Stumbled with this today - It seems unresolved, no?

shr-project, 2023-02-01:
It was resolved a while ago in https://git.openembedded.org/meta-openembedded/commit/?h=kirkstone&id=035d9c61e81b32cb62706cfdbfae08b1bac7ae88

closed 2023-09-12 by aurelihein