openembedded/meta-openembedded #614: fatal: 无法访问 'https://github.com/openembedded/meta-openembedded/'：Empty reply from server

jt0129, 2022-09-26:
meta-openwrt:
fatal: 无法访问 'https://github.com/kraj/meta-openwrt/'：Failed to connect to github.com port 443: Connection timed out
error: Cannot fetch meta-openwrt from git://github.com/kraj/meta-openwrt
Fetching: 100% (11/11), done in 4m59.862s
Garbage collecting: 100% (11/11), done in 0.079s
Fetching: 33% (1/3) meta-clang
fatal: 无法访问 'https://github.com/openembedded/meta-openembedded/'：Empty reply from server

Please help me how can I figure this probelm, thank you!!!

kraj, 2022-09-27:
it must be temporary. Can you retry and use https with github it does not support git protocol anymore.

jt0129, 2022-09-27:
> it must be temporary. Can you retry and use https with github it does not support git protocol anymore.

OMG, thank you for your time! I appreciate your help very much.