openembedded/meta-openembedded #628: protobuf-compiler not including the protobuf compiler (in sdk)

DAdolfsson, 2022-10-21:
Hi,
I recently attempted to assemble a sdk for one of my colleagues but we had problems.

The SDK was successfully built but no protoc compiler was included.
From what i can tell it was not included in the nativesdk-protobuf-compiler package.

I am far from an expert in this, and lots of the recipe goes above my head.
But, to resolve my issues i changed a single line that looked wrong in the recipe:

```diff
diff --git a/meta-oe/recipes-devtools/protobuf/protobuf_3.21.5.bb b/meta-oe/recipes-devtools/protobuf/protobuf_3.21.5.bb
index 0bc9cbedc..e7486857f 100644
--- a/meta-oe/recipes-devtools/protobuf/protobuf_3.21.5.bb
+++ b/meta-oe/recipes-devtools/protobuf/protobuf_3.21.5.bb
@@ -27,7 +27,7 @@ inherit cmake pkgconfig ptest
 PACKAGECONFIG ??= ""
 PACKAGECONFIG:class-native ?= "compiler"
 PACKAGECONFIG[python] = ",,"
-PACKAGECONFIG[compiler] = "-Dprotobuf_BUILD_PROTOC_BINARIES=ON,-Dprotobuf_BUILD_PROTOC_BINARIES=OFF"
+PACKAGECONFIG[compiler] = "-Dprotobuf_BUILD_PROTOC_BINARIES=ON"
 
 EXTRA_OECMAKE += "\
     -Dprotobuf_BUILD_SHARED_LIBS=ON \
```
From my understanding of the precompiler, by adding the same variable twice with different values it will only respect the last one.

Best Regards
Daniel

kraj, 2022-10-21:
Can you test this patch instead ? and report back if it works

```diff
diff --git a/meta-oe/recipes-devtools/protobuf/protobuf_3.21.5.bb b/meta-oe/recipes-devtools/protobuf/protobuf_3.21.5.bb
index 0bc9cbedc0..c8b9158e6c 100644
--- a/meta-oe/recipes-devtools/protobuf/protobuf_3.21.5.bb
+++ b/meta-oe/recipes-devtools/protobuf/protobuf_3.21.5.bb
@@ -26,6 +26,7 @@ inherit cmake pkgconfig ptest

 PACKAGECONFIG ??= ""
 PACKAGECONFIG:class-native ?= "compiler"
+PACKAGECONFIG:class-nativesdk ?= "compiler"
 PACKAGECONFIG[python] = ",,"
 PACKAGECONFIG[compiler] = "-Dprotobuf_BUILD_PROTOC_BINARIES=ON,-Dprotobuf_BUILD_PROTOC_BINARIES=OFF"

```

DAdolfsson, 2022-10-24:
Thank you,

I it appears to have solved the issue for the generated SDK!

I did a clean-all on the proto* packages and the image.. Verified that the package files was gone and rebuilt..

Now the protoc compiler is present

DAdolfsson, 2022-10-24:
Ah.. just saw that you already included it in master.

Thank you!

commit 4d0bf95e (2022-06-06): unbound: upgrade 1.15.0 -> 1.16.0

Changelog:
==========
This release has EDE support, for extended EDNS error reporting,
it fixes unsupported ZONEMD algorithms to load, and has more bug fixes.

The EDE errors can be turned on by 'ede: yes', it is default disabled.
Validation errors and other errors are then reported. If you also want
stale answers for expired responses to have an error code, the option
'ede-serve-expired: yes' can be used.

Features
- Merge PR #604: Add basic support for EDE (RFC8914).

Bug Fixes
- Fix #412: cache invalidation issue with CNAME+A.
- Fix that TCP interface does [... cut]
closed 2022-10-24 by DAdolfsson