openembedded/meta-openembedded #656: nodejs: Add PACKAGECONFIG share to allow other projects to use node as library

ArchanaWind, 2023-01-31:
Hi ,

The mentioned commit  is adding the below snippet to to the Nodejs recipe

# We don't want to cross-compile during target compile,
@@ -99,6 +101,7 @@ do_configure () {
    ./configure --prefix=${prefix} --without-snapshot --shared-openssl \
                --dest-cpu="${@map_nodejs_arch(d.getVar('TARGET_ARCH'), d)}" \
                --dest-os=linux \
+               --libdir=${D}${libdir} \
                ${ARCHFLAGS} \
                ${PACKAGECONFIG_CONFARGS}
 }

In the above code snipped can we use --libdir=${libdir} instead of --libdir=${D}${libdir}. because due to  ${D}${libdir} getting below do_package_QA warnings

WARNING: nodejs-16.18.1-r0 do_package_qa: QA Issue: File /usr/bin/.debug/node in package nodejs-dbg contains reference to TMPDIR [buildpaths]
WARNING: nodejs-16.18.1-r0 do_package_qa: QA Issue: File /usr/include/node/config.gypi in package nodejs-dev contains reference to TMPDIR [buildpaths]
WARNING: nodejs-16.18.1-r0 do_package_qa: QA Issue: File /usr/bin/node in package nodejs contains reference to TMPDIR [buildpaths]
NOTE: Tasks Summary: Attempted 1028 tasks of which 0 didn't need to be rerun and all succeeded.

Could you please suggest can we use --libdir=${libdir} instead of --libdir=${D}${libdir}

Thanks in advance.

kraj, 2023-01-31:
seems fine, please testout the fix

ArchanaWind, 2023-02-01:
Thank you for quick response

closed 2023-02-18 by ArchanaWind

shr-project, 2023-02-18:
This broke do_install with "shared" PACKAGECONFIG enabled as reported on ML, see the fix in https://lists.openembedded.org/g/openembedded-devel/message/101088