openembedded/meta-openembedded #672: Nodejs:-  ./node: No such file or directory

ArchanaWind, 2023-04-04:
Nodejs binary is not working on host machine

I am working adding Nodejs ptest support, to build tests on host it requires nodejs binary.
Steps to reproduce:
$ bitbake nodejs
$ cd build/tmp/work/core2-64-poky-linux/nodejs/18.14.2-r0/node-v18.14.2
$ ./node 
-sh: ./node: No such file or directory

Can any one please confirm is this the expected behaviour , if then how can I build tests on host as it depends on node binary

kraj, 2023-04-04:
can you post output of `readelf -e node`

ArchanaWind, 2023-04-04:
Output of $readelf -e node:

ELF Header:
  Magic:   7f 45 4c 46 02 01 01 03 00 00 00 00 00 00 00 00 
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - GNU
  ABI Version:                       0
  Type:                              DYN (Position-Independent Executable file)
  Machine:                           Advanced Micro Devices X86-64
  Version:                           0x1
  Entry point address:               0x5269a0
  Start of program headers:          64 (bytes into file)
  Start of section headers:          1364510248 (bytes into file)
  Flags:                             0x0
  Size of this header:               64 (bytes)
  Size of program headers:           56 (bytes)
  Number of program headers:         16
  Size of section headers:           64 (bytes)
  Number of section headers:         42
  Section header string table index: 41

Section Headers:
  [Nr] Name              Type             Address           Offset
       Size              EntSize          Flags  Link  Info  Align
  [ 0]                   NULL             0000000000000000  00000000
       0000000000000000  0000000000000000           0     0     0
  [ 1] .interp           PROGBITS         00000000000003c0  000003c0
       000000000000001a  0000000000000000   A       0     0     1
  [ 2] .note.gnu.pr[...] NOTE             00000000000003e0  000003e0
       0000000000000030  0000000000000000   A       0     0     8
  [ 3] .note.gnu.bu[...] NOTE             0000000000000410  00000410
       0000000000000024  0000000000000000   A       0     0     4
  [ 4] .note.ABI-tag     NOTE             0000000000000434  00000434
       0000000000000020  0000000000000000   A       0     0     4
  [ 5] .gnu.hash         GNU_HASH         0000000000000458  00000458
       0000000000034184  0000000000000000   A       6     0     8
  [ 6] .dynsym           DYNSYM           00000000000345e0  000345e0
       00000000000d9f68  0000000000000018   A       7     1     8
  [ 7] .dynstr           STRTAB           000000000010e548  0010e548
       00000000002b9c39  0000000000000000   A       0     0     1
  [ 8] .gnu.version      VERSYM           00000000003c8182  003c8182
       000000000001229e  0000000000000002   A       6     0     2
  [ 9] .gnu.version_r    VERNEED          00000000003da420  003da420
       00000000000002a0  0000000000000000   A       7     7     8
  [10] .rela.dyn         RELA             00000000003da6c0  003da6c0
       00000000001087c8  0000000000000018   A       6     0     8
  [11] .rela.plt         RELA             00000000004e2e88  004e2e88
       00000000000070f8  0000000000000018  AI       6    27     8
  [12] .init             PROGBITS         00000000004ea000  004ea000
       000000000000001b  0000000000000000  AX       0     0     4
  [13] .plt              PROGBITS         00000000004ea020  004ea020
       0000000000004b60  0000000000000010  AX       0     0     16
  [14] .plt.got          PROGBITS         00000000004eeb80  004eeb80
       0000000000000040  0000000000000008  AX       0     0     8
  [15] .text             PROGBITS         00000000004f0000  004f0000
       0000000000fe7160  0000000000000000  AX       0     0     8192
  [16] lpstub            PROGBITS         0000000001600000  01600000
       0000000000000203  0000000000000000  AX       0     0     2097152
  [17] .fini             PROGBITS         0000000001600204  01600204
       000000000000000d  0000000000000000  AX       0     0     4
  [18] .rodata           PROGBITS         0000000001601000  01601000
       00000000006a9c54  0000000000000000   A       0     0     64
  [19] .eh_frame_hdr     PROGBITS         0000000001caac54  01caac54
       000000000005ba04  0000000000000000   A       0     0     4
  [20] .eh_frame         PROGBITS         0000000001d06658  01d06658
       00000000001a1470  0000000000000000   A       0     0     8
  [21] .tdata            PROGBITS         0000000001ea8e78  01ea7e78
       0000000000000004  0000000000000000 WAT       0     0     8
[... 38 lines cut ...]
  [41] .shstrtab         STRTAB           0000000000000000  5154c482
       00000000000001a2  0000000000000000           0     0     1
Key to Flags:
  W (write), A (alloc), X (execute), M (merge), S (strings), I (info),
  L (link order), O (extra OS processing required), G (group), T (TLS),
  C (compressed), x (unknown), o (OS specific), E (exclude),
  R (retain), D (mbind), l (large), p (processor specific)

Program Headers:
  Type           Offset             VirtAddr           PhysAddr
                 FileSiz            MemSiz              Flags  Align
  PHDR           0x0000000000000040 0x0000000000000040 0x0000000000000040
                 0x0000000000000380 0x0000000000000380  R      0x8
  INTERP         0x00000000000003c0 0x00000000000003c0 0x00000000000003c0
                 0x000000000000001a 0x000000000000001a  R      0x1
      [Requesting program interpreter: /lib/ld-linux-x86-64.so.2]
  LOAD           0x0000000000000000 0x0000000000000000 0x0000000000000000
                 0x00000000004e9f80 0x00000000004e9f80  R      0x1000
  LOAD           0x00000000004ea000 0x00000000004ea000 0x00000000004ea000
                 0x0000000000004bc0 0x0000000000004bc0  R E    0x1000
  LOAD           0x00000000004f0000 0x00000000004f0000 0x00000000004f0000
                 0x0000000000fe7160 0x0000000000fe7160  R E    0x2000
  LOAD           0x0000000001600000 0x0000000001600000 0x0000000001600000
                 0x0000000000000211 0x0000000000000211  R E    0x200000
  LOAD           0

ArchanaWind, 2023-04-05:
Hi @kraj 

Could you please comment on it

kraj, 2023-04-05:
hmmm ` /lib/ld-linux-x86-64.so.2` does that exist on your build system ?
is seems it could be missing the right dynamic linker. Can you try

` /lib/ld-linux-x86-64.so.2 ./node`

shr-project, 2023-04-05:
It's not surprising I think, e.g. my gentoo system has loaded in different directory:
/lib64/ld-linux-x86-64.so.2

which doesn't match with qemux86-64 _TARGET_ build with OE, if you want to run node on your host, build and use nodejs-native.

kraj, 2023-04-05:
> It's not surprising I think, e.g. my gentoo system has loaded in different directory: /lib64/ld-linux-x86-64.so.2
> 

ah good point I did not see the node being run is from `core2-64-poky-linux` which is target build of node for a x86_64 target, so yes using nodejs-native is right aproach

> which doesn't match with qemux86-64 _TARGET_ build with OE, if you want to run node on your host, build and use nodejs-native.

ArchanaWind, 2023-04-06:
Thank you for quick response

I have already tried $bitbake nodejs-native, it didn't work for me. let me try it again

ArchanaWind, 2023-04-08:
nodejs-native is creating the binary at build/tmp/work/x86_64-linux/nodejs-native/18.14.2-r0/node-v18.14.2/node but Nodejs recipe is looking for node binary at build/tmp/work/core2-64-poky-linux/nodejs/18.14.2-r0/node-v18.14.2/node and failing to execute node binary while compiling tests
So I have created soft link to ${RECIPE_SYSROOT_NATIVE}/usr/bin/node, with this tests are built successfully.

Could you please confirm is this the right way of doing.

below is the snippet of code to add nodejs ptest

```
--- a/meta-oe/recipes-devtools/nodejs/nodejs_18.14.2.bb
+++ b/meta-oe/recipes-devtools/nodejs/nodejs_18.14.2.bb
@@ -5,7 +5,7 @@ LIC_FILES_CHKSUM = "file://LICENSE;md5=2dff1ccca11e333f1388e34f7e2d1de3"

 CVE_PRODUCT = "nodejs node.js"

-DEPENDS = "openssl file-replacement-native"
+DEPENDS = "openssl file-replacement-native nodejs-native"
 DEPENDS:append:class-target = " qemu-native"
 DEPENDS:append:class-native = " c-ares-native"

@@ -163,23 +163,40 @@ do_compile () {
     install -D ${B}/v8-qemu-wrapper.sh ${B}/out/Release/v8-qemu-wrapper.sh
     oe_runmake BUILDTYPE=Release
 }

+do_compile_ptest() {

+    ln -sf   ${RECIPE_SYSROOT_NATIVE}/usr/bin/node ${B}/node
+    oe_runmake  test
+}
```

kraj, 2026-10-26:
running target binaries on build host is not going to work in a cross compile env like OE/Yocto

closed 2026-10-26 by kraj