openembedded/meta-openembedded #678: How to increase memory in a RISCV32 image?

mikhailramalho, 2023-04-16:
Hi folks,

I managed to get a RISCV32 image built, but qemu seems to ignore `-m` arguments when starting the image, and sets the memory size to 1G.

Is there any way I can increase it to 4G? I'm trying to build clang with swap, but it's quite slow.

kraj, 2023-04-16:
Yes I have seen this too. thats the max size qemu can emulate for RV32 as it seems. Can you ask this question in qemu mailing list and see if this is expected ?

As far as clang is concerned, we can cross build clang to run on RV32. you can just add

`IMAGE_INSTALL += "clang"` to your image. but if you really want to build it natively on rv32 using qemu then it might be possible if we can allocate 8+ GB RAM to run qemu.

mikhailramalho, 2023-04-16:
> Yes I have seen this too. thats the max size qemu can emulate for RV32 as it seems. Can you ask this question in qemu mailing list and see if this is expected ?

Sure, I'll ask them.

> 
> As far as clang is concerned, we can cross build clang to run on RV32. you can just add
> 
> `IMAGE_INSTALL += "clang"` to your image.

I tried that, with the meta-clang layer, however, I'm getting weird crashes when trying to compile llvm's libc with the cross built clang :/ I'm not sure building 

>  but if you really want to build it natively on rv32 using qemu then it might be possible if we can allocate 8+ GB RAM to run qemu.

For 4G+ RAM we would need a PAE kernel, right? I created an 8G swap and clang's compilation doesn't seem to be taking more than 4G at any given time, so hopefully 4G RAM would be enough.

kraj, 2026-10-26:
is it still an issue ?