openembedded/meta-openembedded #681: No possibility to use FILESEXTRAPATHS:prepend with the redis_7.x.x recipe

graugans, 2023-04-24:
My suspicion is that the redis reipe [redis_7.0.10.bb](https://github.com/openembedded/meta-openembedded/blob/master/meta-oe/recipes-extended/redis/redis_7.0.10.bb#L9) prevents to prepend any path to the FILESPATH variable.

Is there any specific reason to manipulate [FILESPATH](https://docs.yoctoproject.org/ref-manual/variables.html?highlight=filespath#term-FILESPATH) directly? This does break the ability to add a custom `redis.conf` in a custom layer. I was not able to use `FILESEXTRAPATHS:prepend := "${THISDIR}/files:"` in my bbappend.

I do use the kirkstone branch but I guess this may also have an impact on any later branch.

```
# Does not have any impact
FILESEXTRAPATHS:prepend := "${THISDIR}/files:"

# This is an ugly hack, but FILESEXTRAPATHS:prepend does not seem to work
# due to the usage of the FILESPATHS variable directly
REDIS_CONFIG_FILES := "${THISDIR}/files"

SRC_URI += " \
            file://redis.conf \
"

do_install:append() {
    install -d ${D}/${sysconfdir}/redis
    install -m 0644 ${REDIS_CONFIG_FILES}/redis.conf ${D}/${sysconfdir}/redis/redis.conf

    if ${@bb.utils.contains('DISTRO_FEATURES', 'systemd', 'true', 'false', d)}; then
        sed -i 's!daemonize yes!# daemonize yes!' ${D}/${sysconfdir}/redis/redis.conf
        sed -i 's!supervised no!supervised systemd!' ${D}/${sysconfdir}/redis/redis.conf
    fi

}
```

When I add this line in the `SRC_URI`:
```
            file://invalid-file \
```
I can verify that the look-up only takes place in the meta-openembedded layer:

```
WARNING: redis-7.0.9-r0 do_fetch: Failed to fetch URL file://invalid-file, attempting MIRRORS if available
ERROR: redis-7.0.9-r0 do_fetch: Fetcher failure: Unable to find file file://invalid-file anywhere. The paths that were searched were:
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/files/poky
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.9/poky
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/poky
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/poky
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/files/o3r-vpu-c2-tx2-nx
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.9/o3r-vpu-c2-tx2-nx
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/o3r-vpu-c2-tx2-nx
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/o3r-vpu-c2-tx2-nx
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/files/tegra186
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.9/tegra186
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/tegra186
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/tegra186
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/files/armv8a
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.9/armv8a
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/armv8a
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/armv8a
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/files/aarch64
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.9/aarch64
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/aarch64
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/aarch64
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/files/tegra
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.9/tegra
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/tegra
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/tegra
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/files/cuda
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.9/cuda
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/cuda
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/cuda
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/files/aarch64
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.9/aarch64
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/aarch64
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/aarch64
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/files/
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.9/
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/
    /data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/
    /data/downloads

```

kraj, 2023-04-24:
There is a priority order for directories for searching for the files as you can see so
` FILESEXTRAPATHS:prepend := "${THISDIR}/files:"`  gets lower priority than  `FILESEXTRAPATHS:prepend := "${THISDIR}/redis-7.0.9/:"`

try using `FILESEXTRAPATHS:prepend := "${THISDIR}/redis-7.0.9/:"` in your bbappend and rename `files` to `redis-7.0.9` in your layer as well.

graugans, 2023-05-04:
I tried both variants:

- `FILESEXTRAPATHS:prepend := "${THISDIR}/redis-7:"`
- `FILESEXTRAPATHS:prepend := "${THISDIR}/redis-7.0.11:"`

But there is no way to get arround the `/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7`

```
FILESPATH="/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/redis-7/poky
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.11/poky
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/poky
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/poky
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/redis-7/o3r-vpu-c2-tx2-nx
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.11/o3r-vpu-c2-tx2-nx
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/o3r-vpu-c2-tx2-nx
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/o3r-vpu-c2-tx2-nx
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/redis-7/tegra186
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.11/tegra186
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/tegra186
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/tegra186
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/redis-7/armv8a
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.11/armv8a
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/armv8a
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/armv8a
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/redis-7/aarch64
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.11/aarch64
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/aarch64
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/aarch64
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/redis-7/tegra
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.11/tegra
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/tegra
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/tegra
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/redis-7/cuda
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.11/cuda
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/cuda
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/cuda
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/redis-7/aarch64
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.11/aarch64
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/aarch64
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/aarch64
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/redis-7/
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.11/
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/"
```

```
FILESEXTRAPATHS="/data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/redis-7.0.11:__default:"
FILESOVERRIDES="aarch64:cuda:tegra:aarch64:armv8a:tegra186:o3r-vpu-c2-tx2-nx:poky
FILESPATH="/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/redis-7.0.11/poky
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.11/poky
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/poky
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/poky
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/redis-7.0.11/o3r-vpu-c2-tx2-nx
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.11/o3r-vpu-c2-tx2-nx
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/o3r-vpu-c2-tx2-nx
[... cut ...]
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/aarch64
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../../recipes-extended/redis/redis-7.0.11/
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis-7.0.11/
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/redis/
/data/YOCTO/o3r/meta-ifm-tegra/work/build/../layers/meta-openembedded/meta-oe/recipes-extended/redis/files/"
```