openembedded/meta-openembedded #701: awsiot install error

nealdeore, 2023-06-02:
I am trying to install awsiot on yocto build for Rpi4b

I added below line to local.conf -
IMAGE_INSTALL_append = "aws-iot-device-sdk-python-v2 python3-pip"

I am facing below error -
ERROR: Nothing RPROVIDES 'aws-iot-device-sdk-python-v2' (but /home/...../poky/meta/recipes-core/images/core-image-minimal.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'aws-iot-device-sdk-python-v2' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['aws-iot-device-sdk-python-v2']
ERROR: Required build target 'core-image-minimal' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-minimal', 'aws-iot-device-sdk-python-v2']

I have a recipe under meta-openembedded/meta-python/receipes-devtools/python named "aws-iot-device-python_1.4.8.bb"

I am new to yocto. Any help will be much appreciated. Thank you

kraj, 2023-06-06:
so find out who provides `aws-iot-device-sdk-python-v2` see https://layers.openembedded.org/layerindex/recipe/120360/

nealdeore, 2023-06-07:
`python3-aws-iot-device-sdk-python_1.4.8.bb' file is under `meta-openembedded/meta-python/recipes-devtool/python`.

kraj, 2023-06-07:
next is to figure out why bitbake drops it from dep tree. Can you just do `bitbake python3-aws-iot-device-sdk-python` and try to find out what it complains about, also check the recipe as it might have some compatibility conditions encoded

nealdeore, 2023-06-07:
tried `bitbake python3-aws-iot-device-sdk-python` task Success with no errors,

kraj, 2023-06-07:
OK then where is `aws-iot-device-sdk-python-v2` coming from ? notice `-v2` change that to `aws-iot-device-sdk-python`

nealdeore, 2023-06-07:
okay.. got below error not on `bitbake aws-iot-device-sdk-python`

ERROR: Nothing PROVIDES 'aws-iot-device-sdk-python'. Close matches:
  nativesdk-python3-aws-iot-device-sdk-python
  python3-aws-iot-device-sdk-python
  python3-aws-iot-device-sdk-python-native

nealdeore, 2023-06-07:
No errors now!! .. Thanks :D

closed 2026-10-26 by kraj