openembedded/meta-openembedded #704: pipewire-alsa missing symbolic links

cody-ps, 2023-06-05:
It's quite possible I am doing something wrong or missing something here but when building an image with IMAGE_INSTALL += "pipewire pipewire-alsa" the configuration files (50-pipewire.conf and 99-pipewire-default.conf) are installed to /usr/share/alsa.conf.d/ but my understanding is that they need to be symbolically linked from /etc/alsa/conf.d to be enabled.

kraj, 2023-06-06:
you also need to add `pipewire-alsa-card-profile` to `IMAGE_INSTALL`

cody-ps, 2023-06-06:
I added pipewire-alsa-card-profile to my custom image but to no success. Is the expectation that I should be seeing the symlinks added to /etc/alsa/conf.d/? I added pipewire-alsa-card-profile, but I still don't see the pipewire device in the "aplay -lL" output or the symlinks in /etc/alsa/conf.d/ linking to 50-pipewire.conf/99-pipewire-default.conf in /usr/share/alsa.conf.d/.

I also noticed that the PACKAGECONFIG entry "pipewire-alsa" was not being set for the main pipewire recipe. Do you know how or what sets this? I was able to manually add this with a .bbappend but it also did not seem to have an effect.

Thanks for your help.

adalessandro, 2025-03-17:
@kraj AFAICS, `pipewire-alsa-card-profile` is not creating the required symlinks.

According to https://github.com/PipeWire/pipewire/blob/master/INSTALL.md#alsa-plugin
```
The plugin will be picked up by alsa when the following files are in /etc/alsa/conf.d/:

/etc/alsa/conf.d/50-pipewire.conf -> /usr/share/alsa/alsa.conf.d/50-pipewire.conf
/etc/alsa/conf.d/99-pipewire-default.conf
```

This is done in the debian pkg as per:
https://salsa.debian.org/utopia-team/pipewire/-/blob/debian/master/debian/pipewire-alsa.links?ref_type=heads

@kraj am I missing something here? If this is really a bug, I could submit a PR if needed. Thanks!

kraj, 2025-03-17:
yeah you might have discovered a bug here seems to me. Since the fix is to create and package the needed symlinks in do_install, would you mind testing and send the patch for review.

adalessandro, 2025-03-18:
> yeah you might have discovered a bug here seems to me. Since the fix is to create and package the needed symlinks in do_install, would you mind testing and send the patch for review.

Sure. Pushed https://github.com/openembedded/meta-openembedded/pull/949

adalessandro, 2025-03-31:
PR https://github.com/openembedded/meta-openembedded/pull/949 has been merged. Although not backported to previous releases, this issue should be resolved now.