openembedded/meta-openembedded #723: stunnel compiling error

kmeng1219, 2023-07-25:
Any suggestion with the stunnel compiling error?

```
ERROR: stunnel-5.69-r0 do_prepare_recipe_sysroot: The file /usr/include/rpcsvc/nis_object.x is installed by both external-linaro-toolchain and libnsl2, aborting
ERROR: stunnel-5.69-r0 do_prepare_recipe_sysroot:
ERROR: stunnel-5.69-r0 do_prepare_recipe_sysroot: Function failed: extend_recipe_sysroot
```

My project root directory has the following folders:
meta-linaro  meta-openembedded  meta-qt5  meta-security  meta-tegra  meta-tx2  poky

libnsl2 is in poky folder.

kraj, 2023-07-25:
I think its best to look at external toolchain you are using. if it should be a provider for libnsl2 then mark the external toolchain to provide it and then prefer it over system libnsl2 so dependent packages can use one copy/provider of this dependency. You can also delete these files from external toolchain recipe so that it can not conflict with system.

kmeng1219, 2023-07-26:
@kraj Thank you so much for your response. Yes there is some conflict between libnsl2 in stunnel (meta-openembedded) and libnsl in external-linaro-toolchain.bb (meta-linara)

Remove the libnsl2 dependency in stunnel may pass the compile. This is definitely not good solution for sure. Any other solution? Is it a way to let stunnel  to use libnsl from meta-openembedded?  I am not very familier with the bitbake.

kmeng1219, 2023-07-27:
Thanks, Kraj. I will add bbappend to fix the conflict.

closed 2026-10-26 by kraj