openembedded/meta-openembedded #730: Recipe msgpack-cpp fails at do_rootfs

MaistreRenard, 2023-08-10:
Hi,

I'm trying to include the `msgpack-cpp` recipe to my image. I'm working with Yocto on the `honister` branch. This is how I'm including the recipe:
```shell
IMAGE_INSTALL:append = "\
    msgpack-c \
    msgpack-cpp \
"
```
This is my error:
```shel
dev-image-1.0-r0 do_rootfs: Unable to install packages. Command '/workdir/yocto/builds/dev-image/work/zynqmp_generic-poky-linux/dev-image/1.0-r0/recipe-sysroot-native/usr/bin/apt-get  install --allow-downgrades --allow-remove-essential --allow-change-held-packages --allow-unauthenticated --no-remove msgpack-c msgpack-cpp ' returned 100:
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package msgpack-cpp
```
Remark: I can build the recipe using the following command `bitbake msgpack-cpp`

kraj, 2023-08-10:
It might be that the default output package is empty so look in the packages-split directory for the names of output packages generated by this recipe and include the name of the one you need on target

MaistreRenard, 2023-08-14:
So the fix would be to allow the output package to be empty with the following instruction ?
`ALLOW_EMPTY:${PN} = "1"`

kraj, 2023-08-14:
or add `msgpack-cpp-dev` in `IMAGE_INSTALL`

MaistreRenard, 2023-08-14:
It's working fine thank you for your help, have a nice day !

closed 2023-08-14 by MaistreRenard