openembedded/meta-openembedded #732: readline no longer a DEPENDS for jack 1.19.22 on the  master branch

large2021, 2023-08-15:
For recipe meta-oe/recipes-multimedia/jack/jack_1.19.22.bb, on the "master" branch, the **DEPENDS** includes "readline".  Since "jack-utils" is no longer being built, this dependency is _obsolete_.

As "readline" is GPLv3, this introduces the need to "DEPENDS:remove" the dependency for those of us who strive to be GPLv3-free.

large2021, 2023-08-15:
Also present on the "mickledore" branch.

kraj, 2023-08-15:
@large2021 thanks for the report. Are you also working on a patch to fix this. ?

large2021, 2023-08-15:
I’d be happy to submit a patch, and am capable of doing so, but am unfamiliar with the openembedded process/requirements.

Please advise.

Kindest,
Kirk Hays

kraj, 2023-08-15:
you can create a patch and send to openembedded-devel@lists.openembedded.org, you can also create a github pull which I can take to the mailing list.

large2021, 2023-08-15:
A patch has been submitted to the mailing list.

kraj, 2023-08-15:
mailing lists are subscriber only, non-subscriber posts needs to be channelled by a moderator, so it may take a while for your post to reach the mailing list distribution.

closed 2023-08-18 by kraj