openembedded/meta-openembedded #740: Does meta-openembedded kirkstone branch will have update for NodeJS considering that nodejs 16 is now EOL ?

aurelihein, 2023-09-12:
Does meta-openembedded kirkstone branch will have update for NodeJS considering that nodejs 16 is now EOL ([NodeJs Release](https://github.com/nodejs/Release)) ?
What is the plan ?
Should we migrate to another meta-openembedded branch while using meta poky kirkstone branch ?
Thank you for your feedback,
Aurelien BOUIN

akuster, 2023-09-12:
There is no plan. Stable branches typically do not update packages. Its not uncommon for packages to reach EOL will they are present in a Stable branch.   If you need a newer version I suggest either including it in your own layer or submit patches to https://git.yoctoproject.org/meta-lts-mixins so a broader community may contribute.    Using mickledore may be an option but it will EOL as soon as Nanbield is released.

BR,
Armin

aurelihein, 2023-09-12:
Thank you very much @akuster for you answer and suggestions
Best regards,
Aurelien BOUIN

closed 2023-09-12 by aurelihein