openembedded/meta-openembedded #743: Unable to build nodejs-native

paletteguy, 2023-10-02:
Building nodejs-native on kirkstone and mickledore fails with this error:

build/tmp/hosttools/ld: .//Release/obj.target/tools/v8_gypfiles/libv8_initializers.a: error adding symbols: malformed archive
| collect2: error: ld returned 1 exit status
| rm cd3612620867e7ead4ace8fa0a5fd4e7a74528f5.intermediatemake[1]: *** [tools/v8_gypfiles/mksnapshot.target.mk:198: Release/mksnapshot] Error 1
|  a66092d7422de8d53257377725a0e74d1bec44d5.intermediate 8d65583cbda2e9df30d961a1eba822d938bb57dd.intermediate
| make: *** [Makefile:134: node] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.

Build is performed with -j1 to avoid memory issue.

Host: is Ubuntu 22.04 (LinuxThinkPad-P73 5.15.0-84-generic #93-Ubuntu SMP Tue Sep 5 17:16:10 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux)

Building for target arm64 works fine.

kraj, 2023-10-04:
malformed archive, means something crashed perhaps during build. Check if build system dumped some cores and what those are. Its required to have 16G minimal RAM for building yocto without problems.

paletteguy, 2023-10-05:
I've got 32 GB of memory 500 GB left on drive and I'm compiling with PARALLEL_MAKE = "-j1" as append to receipe for nodejs

kraj, 2023-10-05:
mksnapshot needs a bit of RAM but 32gb should be fine. Can you see if backporting 33c96b7ed85045e3d5bc169908fc25c03efcf9a5 helps.

paletteguy, 2023-10-06:
Thanks for support. 

The root-cause was a homebrew installation causing the linking to malformed. After uninstalling Homebrew I had no issue compiling nodejs-native.

closed 2023-10-06 by paletteguy