openembedded/meta-openembedded #744: please remove python3-botocore and python3-boto3

thomas-roos, 2023-10-05:
Hi,
we are maintaining meta-aws and those two recipes were recently 
https://github.com/openembedded/meta-openembedded/blob/master/meta-python/recipes-devtools/python/python3-botocore_1.31.57.bb
https://github.com/openembedded/meta-openembedded/blob/master/meta-python/recipes-devtools/python/python3-boto3_1.28.57.bb
added causing issues with https://bugzilla.yoctoproject.org/show_bug.cgi?id=15223
Discussed with Richard Purdie and he said people should just use ours - you might have not seen this.
We do ptests and daily upgrades for those recipes.
-> https://github.com/aws4embeddedlinux/meta-aws/tree/master/recipes-devtools/python
Thank you!
Cheers, Thomas

kraj, 2023-10-05:
yeah I am fine with it, Please send a pull to delete them with reasons as discussed.

kraj, 2023-10-11:
done

closed 2023-10-11 by kraj