openembedded/meta-openembedded #747: Cannot find nlohmann-json header files

abs3n, 2023-10-13:
Hi, I was able to successfully install `meta-oe/recipes-devtools/nlohmann-json` by adding `IMAGE_INSTALL:append = " nlohmann-json"` in the `bblayers.conf` file , but for some reason, the rootfs image does not have any header file under `/usr/include/nlohmann/*` when I checked by booting in qemu. Tried using ` find..  -name '*json.hpp*' ` , still no luck. 

Any leads would be appreciated!

shr-project, 2023-10-13:
IIRC it's header only recipe, so all the files are installed in PN-dev package and PN is empty.

abs3n, 2023-10-13:
Yes it is a header only, but apologies for my rookie knowledge in Yocto- when you say PN-dev, what exactly do you mean? I mean, for me to run a program that uses `nlohmann/json.hpp`, the header files need to exist in a location. So I was expecting the files to end up on rootfs under the standard header file path- `/usr/include/*`

abs3n, 2023-10-13:
Never mind, just a freudian slip. Header only's are statically linked anyway, so I guess I should be fine with running my binary and do not even need to install `nlohmann-json` on target anyway.

closed 2023-10-13 by abs3n