openembedded/meta-openembedded #752: libgpiod_1.6.3 does not install the binaries

FlaviuNistor, 2023-10-24:
Hello,

I used libgpiod_1.4.3 (dunfell) up to now and having libgpiod appended in IMAGE_INSTALL was enough to have the binaries (installed) for the gpio tools part of the target image.  
Switching to 1.6.3, now I don't have them installed any more. I can see that yocto does compile them and the *.ipk are generated, but they are not installed in the target image no more (not part of /usr/bin/).
I can't find any info that I need to do more them this.

lms-ts, 2023-10-24:
https://github.com/openembedded/meta-openembedded/blob/langdale/meta-oe/recipes-support/libgpiod/libgpiod_1.6.3.bb

The binaries are put into the tools package by the following line:

FILES:${PN}-tools = " \
    ${bindir}/gpiodetect \
    ...
"

where ${bindir} is /usr/bin.

Try to explicitly add the tools package to the image via

IMAGE_INSTALL:append = " libgpiod-tools"

FlaviuNistor, 2023-10-24:
Thanks for the hint. I will close the issue now.