openembedded/meta-openembedded #755: build error for graphviz

bchoineubility, 2023-10-26:
Hi,

I faced build error for graphviz as following.

```
ERROR: graphviz-2.50.0-r0 do_prepare_recipe_sysroot: Error executing a python function in exec_func_python() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_func_python() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:extend_recipe_sysroot(d)
     0003:
File: '/home/bchoi/nvidia-yocto-bsp/sources/poky/meta/classes/staging.bbclass', lineno: 626, function: extend_recipe_sysroot
     0622:    for f in fixme:
     0623:        staging_processfixme(fixme[f], f, recipesysroot, recipesysrootnative, d)
     0624:
     0625:    for p in postinsts:
 *** 0626:        subprocess.check_output(p, shell=True, stderr=subprocess.STDOUT)
     0627:
     0628:    for dep in manifests:
     0629:        c = setscenedeps[dep][0]
     0630:        os.symlink(manifests[dep], depdir + "/" + c + ".complete")
File: '/usr/lib/python3.8/subprocess.py', lineno: 415, function: check_output
     0411:        else:
     0412:            empty = b''
     0413:        kwargs['input'] = empty
     0414:
 *** 0415:    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
     0416:               **kwargs).stdout
     0417:
     0418:
     0419:class CompletedProcess(object):
File: '/usr/lib/python3.8/subprocess.py', lineno: 516, function: run
     0512:            # We don't call process.wait() as .__exit__ does that for us.
     0513:            raise
     0514:        retcode = process.poll()
     0515:        if check and retcode:
 *** 0516:            raise CalledProcessError(retcode, process.args,
     0517:                                     output=stdout, stderr=stderr)
     0518:    return CompletedProcess(process.args, retcode, stdout, stderr)
     0519:
     0520:
Exception: subprocess.CalledProcessError: Command '/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-poky-linux/graphviz/2.50.0-r0/recipe-sysroot-native/usr/bin/postinst-graphviz-native' returned non-zero exit status 126.

Subprocess output:
/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-poky-linux/graphviz/2.50.0-r0/recipe-sysroot-native/usr/bin/postinst-graphviz-native: 3: dot: Permission denied

ERROR: Logfile of failure stored in: /home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-poky-linux/graphviz/2.50.0-r0/temp/log.do_prepare_recipe_sysroot.2376033

```

My build env is 
```
<?xml version="1.0" encoding="UTF-8"?>
<manifest>

  <default sync-j="8"/>

  <remote fetch="https://github.com/OE4T" name="OE4T"/>
  <remote fetch="https://github.com/OSSystems" name="OSSystems"/>
  <!--remote fetch="https://code.qt.io/yocto"  name="QT5"/-->
  <remote fetch="https://github.com/meta-qt5"  name="meta-qt5"/>
  <remote fetch="https://github.com/TimesysGit"  name="Timesys"/>
  <remote fetch="https://github.com/kraj" name="clang"/>
  <remote fetch="https://github.com/openembedded" name="oe"/>
  <!--remote fetch="https://github.com/ros" name="ros"/-->
  <remote fetch="https://github.com/moto-timo" name="ros"/>
  <remote fetch="https://git.yoctoproject.org/git" name="yocto"/>
  <!--remote fetch="https://github.com/bchoineubility" name="bchoineubility"/-->
  <remote fetch="git@github.com:bchoineubility" name="bchoineubility"/>
[... 96 lines cut ...]
                        python3-colcon-python-setup-py python3-colcon-recursive-crawl \
                        python3-colcon-ros python3-colcon-test-result \
                        "

IMAGE_INSTALL:append = " python3-humanfriendly python3-colcon-ros python-cmake-module cmake gcc g++ gcc-symlinks g++-symlinks \
                        ament-cmake-libraries-dev ament-cmake-export-definitions-dev ament-cmake-export-include-directories-dev \
                        foonathan-memory-staticdev \
                        "
IMAGE_INSTALL:append = " ament-cmake-export-definitions \
                          ament-cmake-export-dependencies \
                          ament-cmake-export-include-directories \
                          ament-cmake-export-interfaces \
                          ament-cmake-export-libraries \
                          ament-cmake-export-link-flags \
                          ament-cmake-export-targets \
"

IMAGE_INSTALL:append = " gtest-vendor-dev clang ament-cpplint"

TOOLCHAIN_HOST_TASK:append = " nativesdk-python3-colcon-common-extensions nativesdk-python3-colcon-core"

```

BR,
Mark

bchoineubility, 2023-10-26:
I  changed permission for dot manually.

```
bchoi@nvidia-build:~/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-poky-linux/graphviz/2.50.0-r0/recipe-sysroot-native/usr/bin$ ls -al dot
-rw-r--r-- 3 bchoi bchoi 14360 10월 26 12:54 dot
bchoi@nvidia-build:~/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-poky-linux/graphviz/2.50.0-r0/recipe-sysroot-native/usr/bin$ chmod 777 dot
bchoi@nvidia-build:~/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-poky-linux/graphviz/2.50.0-r0/recipe-sysroot-native/usr/bin$ ll dot
-rwxrwxrwx 3 bchoi bchoi 14360 10월 26 12:54 dot*
bchoi@nvidia-build:~/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-poky-linux/graphviz/2.50.0-r0/recipe-sysroot-native/usr/bin$
```

Thanks.

bchoineubility, 2023-10-26:
This patch would be required.

[0001-Change-access-right_dot.patch](https://github.com/openembedded/meta-openembedded/files/13173081/0001-Change-access-right_dot.patch)

Can you look into it?

BR,
Mark

kraj, 2026-10-26:
Can you still reproduce this on latest master branch ?