openembedded/meta-openembedded #757: Issue with POCO installation

abs3n, 2023-11-07:
Hi, can someone tell me why I am getting the below errors, despite adding DEPENDS = "poco" in my package recipe?

```
ERROR: scanner-v1.0-r0 do_package_qa: QA Issue: /usr/bin/demo contained in package scanner requires libPocoNet.so.95()(64bit), but no providers found in RDEPENDS:scanner? [file-rdeps]
ERROR: scanner-v1.0-r0 do_package_qa: QA Issue: /usr/bin/demo contained in package scanner requires libPocoFoundation.so.95()(64bit), but no providers found in RDEPENDS:scanner? [file-rdeps]
ERROR: scanner-v1.0-r0 do_package_qa: QA Issue: /usr/bin/demo contained in package scanner requires libPocoZip.so.95()(64bit), but no providers found in RDEPENDS:scanner? [file-rdeps]
```

This is from my recipe- 

```
DEPENDS = "poco" 

do_install() {
    install -d ${D}/usr/bin
    install -m 0755 ${THISDIR}/files/demo ${D}/usr/bin/
}
```

abs3n, 2023-11-08:
Ok, found the root cause. Seems like `meta-oe` is still fetching Poco `1.12.4` while I compiled with `1.12.5`.  How often do you maintain/upgrade POCO recipes?

kraj, 2023-11-08:
It depends on when someone sends a patch to bump the version

kraj, 2026-10-26:
its at 1.15.4 in master

closed 2026-10-26 by kraj