openembedded/meta-openembedded #772: libgpiod cxx bindings license LGPLv3+

bshm, 2024-01-07:
Dear maintainers, @brgl,

I noticed that the the libgpiod cxx bindings have been re-licensed from LGPLv2+ to LGPLv3+ starting from version 2.0.
https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/commit/bindings/cxx/gpiod.hpp?id=ea84f882d5d39b734a5756b3a6950da4896b9a0a

I think this should be in reflected in the LICENSE line of 
[libgpiod_2.1.bb](https://github.com/openembedded/meta-openembedded/blob/master/meta-oe/recipes-support/libgpiod/libgpiod_2.1.bb#L3) which currently lists only
```
LICENSE = "GPL-2.0-or-later & LGPL-2.1-or-later & CC-BY-SA-4.0"
```

brgl, 2024-01-08:
Which part of `LGPL-2.1-or-later` is not matched by `LGPL-3.0` exactly?

bshm, 2024-01-08:
In general I think the LICENSE field in yocto recipes is what SPDX calls a license expression:
https://spdx.github.io/spdx-spec/v2-draft/SPDX-license-expressions/
> [...] Another example is when a set of licenses is needed to represent a binary program constructed by compiling and linking two (or more) different source files each governed by different licenses (e.g., LGPL-2.1-only AND BSD-3-Clause).

My concrete understanding of the semantics of the LICENSE field in libgpiod is:

> Libgpiod source files consist of 
> files that a user is allowed to use under the terms of the GPL-2.0 license or any later version of the GPL license and
> files that a user is  allowed to under the terms of the LGPL-2.1 or any later version of the LGPL license and
> files that a user is allowed to under the terms of the  CC-BY-SA-4.0 license.

The libgpiod C++ bindings are now re-licensed LGPL-3, that means a user of the library does _not_ have the choice to use those under LGPL-2.1 anymore -- and imho the package's LICENSE field should reflect that.

This results in practice the LICENSE field cannot be used by yocto anymore to identify incompatible licenses or to generate concise SBOMs for the build artifacts.

I think this could be solved the following way, setting the LICENSE for the libgpiodcxx subpackage setting to LGPL-3.0, something like this for libgpiod-2.1.bb
```
LICENSE = "GPL-2.0-or-later & LGPL-2.1-or-later & CC-BY-SA-4.0 & LGPL-3.0"
LICENSE:${PN} = "GPL-2.0-or-later & LGPL-2.1-or-later & CC-BY-SA-4.0"
LICENSE:libgpiodcxx = "LGPL-3.0"
```

brgl, 2024-01-09:
I see. Yes it makes sense but if you want to be 100% strict then there would be non LGPL-3.0 files too in the libgpiod package. Do you want to send a patch to the list?

bshm, 2024-01-11:
>  if you want to be 100% strict then there would be non LGPL-3.0 files too in the libgpiod package
I don't fully understand. Did you mean to say libgpiodcxx package?

> Do you want to send a patch to the list?
Honestly I don't have much experience with sending patches and would prefer creating a PR here. But if this usual openembedded development workflow, yes.

kraj, 2026-10-26:
it moved back to LGPL-2.0-later in https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/commit/bindings/cxx/gpiod.hpp?id=76ecc337c519d65c16e331343c9402f176eead5a

There are also improvements in recipe via 01ec9c3353f0980f229fcd9cf72eebbd43d4384c

closed 2026-10-26 by kraj