openembedded/meta-openembedded #776: python twisted 23.10

jbleyel, 2024-01-23:
Please add python3-twisted_23.10.0.bb to master

Thanks

gudnimg, 2024-04-01:
Just adding that 24.3.0 has been released meanwhile: https://github.com/twisted/twisted/releases

kraj, 2024-04-01:
Thanks, I think we need someone to draft a PR with upgrade PR

gudnimg, 2024-04-13:
I've been looking into what needs to change to upgrade the recipe the past weeks. So far it builds fine on my end using Kirkstone. But I think it is time to do some cleanup in the recipe.

`FILES:${PN}-core` for example is referencing non-existing files like `${libdir}/${PYTHON_DIR}/site-packages/twisted/python/timeoutqueue.py* \` which was removed in Twisted 10.2.0 :)  I would have thought Bitbake raises an error, but apparently not.

gudnimg, 2024-04-13:
In case no one updates Twisted before I submit a patch I will try to do this with two patches:

* Patch 1: Cleanup the current Twisted 22.10.0 recipe by mainly removing pointless `FILES` entries
* Patch 2: only add necessary changes to update Twisted from 22.10.0 to 24.3.0

I have the changes already, just need to separate them into two commits. Will probably finish it tomorrow or over the next week.

gudnimg, 2024-04-18:
Just sent the final patch for review which upgrades twisted to 24.3.0 :)

closed 2026-10-26 by kraj