openembedded/meta-openembedded #777: Testcases are missing under kernel-selftest.bb

cshilwant, 2024-01-25:
Why are we not adding all kernel selftests in [kernel-selftest.bb](https://git.openembedded.org/meta-openembedded/tree/meta-oe/recipes-kernel/kernel-selftest/kernel-selftest.bb?h=kirkstone)?

Initial commit on kirkstone branch suggests plan was to add more testcases in future - https://github.com/openembedded/meta-openembedded/commit/146de1dfdac3e640af1b082b3a0c8bcf4ee8dad3 which never happened!

kraj, 2024-01-25:
someone needs to do the work and propose patches and clearly no one has so far.

cshilwant, 2024-01-25:
@kraj Cool, I will send out a patch adding certain tests if that helps!

cshilwant, 2024-01-26:
@kraj Created following PR's on master and kirkstone to address to add certain self-tests,

- https://github.com/openembedded/meta-openembedded/pull/778
- https://github.com/openembedded/meta-openembedded/pull/779

cshilwant, 2024-02-08:
@kraj  Any reasons [kernel-selftest: Add few more testcases #779](https://github.com/openembedded/meta-openembedded/pull/779) is still open?

kraj, 2024-02-08:
yeah go ahead and close it referring to commits

cshilwant, 2024-02-11:
@kraj Sure will close the issue once #779 is integrated in kirkstone branch by @akuster.

kraj, 2026-10-26:
Closing as kirkstone is EOL

closed 2026-10-26 by kraj