openembedded/meta-openembedded #800: VLC with master branch compilation

VaginAY, 2024-03-18:
I tried to compile vlc with the latest version of the meta-openembedded layer (scacrthgap) and got errors. I provide the log file below.

```
meta_engine/taglib.cpp:185:32: error: 'null' is not a member of 'TagLib::ByteVector'
  185 |             return ByteVector::null;
      |                                ^~~~
```

and 

```
libtool: link: echo "{ global:" > .libs/libpacketizer_flac_plugin.ver
libtool: link:  cat .libs/libpacketizer_flac_plugin.exp | sed -e "s/\(.*\)/\1;/" >> .libs/libpacketizer_flac_plugin.ver
meta_engine/taglib.cpp: In function 'void ReadMetaFromMP4(TagLib::MP4::Tag*, demux_meta_t*, vlc_meta_t*)':
meta_engine/taglib.cpp:781:14: error: 'class TagLib::MP4::Tag' has no member named 'itemListMap'; did you mean 'itemMap'?
  781 |     if( tag->itemListMap().contains(keyName) )                                               \
      |              ^~~~~~~~~~~
meta_engine/taglib.cpp:793:5: note: in expansion of macro 'SET'
  793 |     SET("----:com.apple.iTunes:MusicBrainz Track Id", TrackID );
      |     ^~~
meta_engine/taglib.cpp:783:21: error: 'class TagLib::MP4::Tag' has no member named 'itemListMap'; did you mean 'itemMap'?
  783 |         list = tag->itemListMap()[keyName];                                                  \
      |                     ^~~~~~~~~~~
meta_engine/taglib.cpp:793:5: note: in expansion of macro 'SET'
  793 |     SET("----:com.apple.iTunes:MusicBrainz Track Id", TrackID );
      |     ^~~
meta_engine/taglib.cpp:787:14: error: 'class TagLib::MP4::Tag' has no member named 'itemListMap'; did you mean 'itemMap'?
  787 |     if( tag->itemListMap().contains(keyName) )                                  \
      |              ^~~~~~~~~~~
meta_engine/taglib.cpp:794:5: note: in expansion of macro 'SET_EXTRA'
  794 |     SET_EXTRA("----:com.apple.iTunes:MusicBrainz Album Id", VLC_META_EXTRA_MB_ALBUMID );
      |     ^~~~~~~~~
meta_engine/taglib.cpp:789:21: error: 'class TagLib::MP4::Tag' has no member named 'itemListMap'; did you mean 'itemMap'?
  789 |         list = tag->itemListMap()[keyName];                                     \
      |                     ^~~~~~~~~~~
meta_engine/taglib.cpp:794:5: note: in expansion of macro 'SET_EXTRA'
  794 |     SET_EXTRA("----:com.apple.iTunes:MusicBrainz Album Id", VLC_META_EXTRA_MB_ALBUMID );
      |     ^~~~~~~~~
meta_engine/taglib.cpp:799:14: error: 'class TagLib::MP4::Tag' has no member named 'itemListMap'; did you mean 'itemMap'?
  799 |     if( tag->itemListMap().contains("covr") )
      |              ^~~~~~~~~~~
      |              itemMap
meta_engine/taglib.cpp:801:39: error: 'class TagLib::MP4::Tag' has no member named 'itemListMap'; did you mean 'itemMap'?
  801 |         MP4::CoverArtList list = tag->itemListMap()["covr"].toCoverArtList();
      |                                       ^~~~~~~~~~~
      |                                       itemMap
meta_engine/taglib.cpp: In function 'int ReadMeta(vlc_object_t*)':
meta_engine/taglib.cpp:897:23: error: 'class TagLib::String' has no member named 'isNull'
  897 |     if( !p_tag->tag().isNull() && !p_tag->tag().isEmpty() )                    \
      |                       ^~~~~~
meta_engine/taglib.cpp:907:5: note: in expansion of macro 'SET'
  907 |     SET( title, Title );
      |     ^~~
meta_engine/taglib.cpp:897:23: error: 'class TagLib::String' has no member named 'isNull'
  897 |     if( !p_tag->tag().isNull() && !p_tag->tag().isEmpty() )                    \
      |                       ^~~~~~
meta_engine/taglib.cpp:908:5: note: in expansion of macro 'SET'
  908 |     SET( artist, Artist );
      |     ^~~
meta_engine/taglib.cpp:897:23: error: 'class TagLib::String' has no member named 'isNull'
  897 |     if( !p_tag->tag().isNull() && !p_tag->tag().isEmpty() )                    \
      |                       ^~~~~~
meta_engine/taglib.cpp:909:5: note: in expansion of macro 'SET'
  909 |     SET( album, Album );
      |     ^~~
meta_engine/taglib.cpp:897:23: error: 'class TagLib::String' has no member named 'isNull'
  897 |     if( !p_tag->tag().isNull() && !p_tag->tag().isEmpty() )                    \
      |                       ^~~~~~
meta_engine/taglib.cpp:910:5: note: in expansion of macro 'SET'
  910 |     SET( comment, Description );
      |     ^~~
meta_engine/taglib.cpp:897:23: error: 'class TagLib::String' has no member named 'isNull'
  897 |     if( !p_tag->tag().isNull() && !p_tag->tag().isEmpty() )                    \
      |                       ^~~~~~
meta_engine/taglib.cpp:911:5: note: in expansion of macro 'SET'
  911 |     SET( genre, Genre );
      |     ^~~
libtool: link:  echo "local: *; };" >> .libs/libpacketizer_flac_plugin.ver
meta_engine/taglib.cpp:974:45: error: invalid conversion from 'TagLib::Tag*' to 'TagLib::ID3v2::Tag*' [-fpermissive]
  974 |             ReadMetaFromId3v2( riff_wav->tag(), p_demux_meta, p_meta );
      |                                ~~~~~~~~~~~~~^~
      |                                             |
      |                                             TagLib::Tag*
```

kraj, 2024-03-18:
I think we need more backports from master branch of vlc to support taglib 2.12. I proposed something which builds it with clang compiler but its not all, there are more fixes [1] in master branch that might be needed on 3.x

[1] https://mailman.videolan.org/pipermail/vlc-devel/2024-March/143639.html

VaginAY, 2024-03-19:
Find, that compilation is good, if i take last changes from master. Sorry for the false alarm

closed 2026-10-26 by kraj