openembedded/meta-openembedded #805: Question: `python3-twisted` on Kirkstone is missing two runtime dependencies

gudnimg, 2024-04-02:
I noticed when upgrading a project to Kirkstone that Twisted 22.2.0 is missing two runtime dependencies which were later added with these following commits, but haven't been backported:

* https://git.openembedded.org/meta-openembedded/commit/meta-python/recipes-devtools/python/python3-twisted_22.10.0.bb?h=mickledore&id=78386ad01d963b84749c28d6d0c312405f778287
* https://git.openembedded.org/meta-openembedded/commit/meta-python/recipes-devtools/python/python3-twisted_22.10.0.bb?h=mickledore&id=0cd4da100fc2874c233155c88be2e6dc376567a0

I currently have a `bbappend` file in my project to resolve this, but it would be nice to fix the Kirkstone LTS branch. 

Adding `python3-typing-extensions` and `python3-asyncio` to `RDEPENDS:${PN}-core` should resolve the issue. Minimum requirement for `python3-typing-extensions` is 3.6.5, and the OpenEmbedded recipe provides version 3.10.0.0 ✔️ 

I can try submit a patch for this, but wanted to ask first if a fix like this is considered for merging? This is merely just to fix the recipe upstream.

kraj, 2024-04-03:
@gudnimg thanks for report and yes this is a good fix for stable branch, please check with `master` branch first and if its not fixed there, then propose it for master, if its already fixed in master via another commit then backport that to stable branch, if not backportable, then modify the fix suitable for stable branch

gudnimg, 2024-04-03:
> if its already fixed in master via another commit then backport that to stable branch

These two commits are in master:

* https://github.com/openembedded/meta-openembedded/commit/0cd4da100fc2874c233155c88be2e6dc376567a0
* https://github.com/openembedded/meta-openembedded/commit/78386ad01d963b84749c28d6d0c312405f778287

I’m not sure about the best approach to send this via the mailing list (still learning 😀). Do I just cherry pick these on a local kirkstone branch and sign the commits like I would regular commits?

gudnimg, 2024-04-03:
On an unrelated note, I would love to backport these two commits as well (they’re from me BTW). They are not critical, but cleaning dependencies is really importart for my application to minimize image size (pip is ~10MB)

* https://github.com/openembedded/meta-openembedded/commit/28defafdc5ff3d7254781d04df718065681f7769
* https://github.com/openembedded/meta-openembedded/commit/55017a608e5c8735b983ae06f206ce559fa6afe4

But I will focus on twisted first since it causes an actual error. 😊

closed 2024-04-03 by gudnimg