openembedded/meta-openembedded #822: Broken php-fpm systemd service in kirkstone and mickledore

bence98, 2024-05-03:
Commit 4cefe01 ("php-fpm: fix systemd") fixes php-fpm's systemd service. However, this was not backported to kirkstone and mickledore.

bence98, 2024-07-18:
Since then, mickledore and nanbield seem to have gone EOL, and this issue was never fixed. Is there any plans to fix this in kirkstone, which is supposed to be LTS until 2026, or are we just supposed to stick the support up our own arse?

shr-project, 2024-07-18:
What about reading README and sending backport request to ML?

shr-project, 2024-07-18:
You should send the patch you want to backport to ML with [kirkstone] in subject like README says https://git.openembedded.org/meta-openembedded/tree/meta-oe/README?h=kirkstone#n21

bence98, 2024-08-06:
Fixed in kirkstone as of 69d1121922be.

closed 2024-08-06 by bence98