openembedded/meta-openembedded #826: scarthgap: auditd recipe broken when systemd is enabled

embetrix, 2024-05-17:
```
ERROR: audit-4.0.1-r0 do_package: Didn't find service unit 'auditd.service', specified in SYSTEMD_SERVICE:auditd. 
ERROR: Logfile of failure stored in: /workspaces/meta-ozyx/build/tmp/work/core2-32-ozyx-linux/audit/4.0.1/temp/log.do_package.2949105
ERROR: Task (/workspaces/meta-example/build/../layers/meta-openembedded/meta-oe/recipes-security/audit/audit_4.0.1.bb:do_package) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6613 tasks of which 4859 didn't need to be rerun and 1 failed.
```

embetrix, 2024-05-17:
the unit files install step is missing: 
```
do_install:append() {

	install -d ${D}${systemd_unitdir}/system
	install -m 0644 ${B}/init.d/auditd.service      ${D}${systemd_unitdir}/system/
	install -m 0644 ${B}/init.d/audit-rules.service ${D}${systemd_unitdir}/system/

}
```

pityhelplessnessandfat, 2024-06-28:
The following modifications need to be add. If a `DISTRO_FEATURES` of project contains `sysvinit` and `systemd`, `${D}${libdir}/systemd` will be deleted. Even if `DISTRO_FEATURES` only contains `sysvinit`, `${D}` still is deleted at beginning of next task.
```
     if ${@bb.utils.contains('DISTRO_FEATURES', 'sysvinit', 'true', 'false', d)}; then
         install -D -m 0755 ${WORKDIR}/auditd ${D}/etc/init.d/auditd
-        rm -rf ${D}${libdir}/systemd
     fi

```

akuster, 2024-10-11:
I'll be taking a look into this. Will check master too

akuster, 2024-10-11:
Yeah, master fails too

akuster, 2024-10-13:
patch sent to mailing list.

OldManYellsAtCloud, 2025-03-14:
@sgorham Feel free to send a patch/PR. Backports happen when someone cares about it strong enough to make it happen

kraj, 2026-10-26:
fixed with ff87862031a66002189206d3855ad94d44b8636e on scarthgap

closed 2026-10-26 by kraj