openembedded/meta-openembedded #832: Configuration errors for libblockdev

pasweistorz, 2024-06-12:
We use the following `PACKAGECONFIG` for libblockdev.

```
PACKAGECONFIG:pn-libblockdev = "crypto fs mdraid parted"
```

This leads to the following errors during configuration:

```
| Checking header keyutils.h existence and usability.../tmp/d9dhL.c:1:10: fatal error: keyutils.h: No such file or directory
|     1 | #include <keyutils.h>
|       |          ^~~~~~~~~~~~
| compilation terminated.
| no
| checking for fdisk >= 2.31.0... yes
| checking for uuid... yes
| checking for uuid... yes
| checking for mount >= 2.23.0... yes
| checking for ext2fs e2p... no
| checking for blkid >= 2.23.0... yes
| checking for bytesize >= 0.1... yes
| checking for libnvme >= 1.3... yes
| 
| *** Libblockdev encountered the following issues during configuration:
| 
| keyutils.h not available
| No package 'ext2fs' found
| No package 'e2p' found
| 
| *** Libblockdev will not successfully build without these missing dependencies
```

It seems that e2fsprogs and keyutils are missing as build dependencies. Adding e2fsprogs to the build dependencies in case `PACKAGECONFIG` contains `fs` and adding keyutils to the build dependencies in case `PACKAGECONFIG` contains `crypto` fixes the issue:

```
PACKAGECONFIG[fs] = "--with-fs, --without-fs, e2fsprogs util-linux"
PACKAGECONFIG[crypto] = "--with-crypto,--without-crypto,cryptsetup keyutils nss volume-key"
```

kraj, 2024-06-12:
@pasweistorz thanks for the report. It seems that you have also identified the fix by adding missing deps. It would be great if you can turn that into a patch and eventual pull request

pasweistorz, 2024-06-13:
@kraj Thanks for your quick response. I just sent a pull request to the mailing list. Please have a look at it.

kraj, 2026-10-26:
fixed

closed 2026-10-26 by kraj