openembedded/meta-openembedded #837: [meta-oe][scarthgap] openbox crashes on alt+tab with fullscreen app probably caused by newer glib-2.0

FederAndInk, 2024-06-26:
When alt tabbing from a fullscreen app (chromium), openbox crashes, this is apparently due to glib-2.0 upgrade (>= 2.76)

found related bug on arch: https://bugs.archlinux.org/task/77853

the commit on the openbox repo: https://github.com/Mikachu/openbox/commit/9ed6fdd71890c5cc43747f105382d5677e5d37e7

I haven't tried the patch yet, will report back when tried

kraj, 2024-06-26:
Thanks for report @FederAndInk if this fixes the problem, it would be good to send a patch to openbox recipe to apply this patch

FederAndInk, 2024-06-26:
Tried it, the patch fixes this issue, unfortunately I don't have the time to send the patch for now

ycongal-smile, 2024-06-27:
@kraj @FederAndInk We will handle it from here and send a patch :)

ycongal-smile, 2024-07-11:
Patch sent : https://lists.openembedded.org/g/openembedded-devel/topic/meta_oe_patch_openbox_fix/107168578

ycongal-smile, 2024-07-15:
Merged in master : https://git.openembedded.org/meta-openembedded/commit/?id=85132c16210d1bdf83e8be5b3da4f430ce7b4b91

kraj, 2024-07-15:
Thanks @ycongal-smile we can close this issue. Please re-open if something more is needed on this.

closed 2024-07-15 by kraj