openembedded/meta-openembedded #839: libiio bitbake file incorrect addition of PACKAGES due to use of "=+" instead of "+="

tjpsantiago16, 2024-06-28:
https://github.com/openembedded/meta-openembedded/blob/0c4079fc28cc36cb2dbfe48093f2cf64106f8b5d/meta-oe/recipes-support/libiio/libiio_0.25.bb#L46

kraj, 2024-06-28:
Please describe the issue with some more detail, since += or =+ are valid ways of adding to a bitbake variable.

tjpsantiago, 2024-06-29:
Hi @kraj ,

sorry. now that you mention it, I think we need to check again on our side why on our side this =+ is not working as how we expected it to be. libiio-iiod, libiio-tests and libiio-python3 was not being installed in our final image.

do you know any way to debug this problem? currently we are only debugging by checking if iio-info and other expected binaries are available in the final image.

closed 2024-06-29 by tjpsantiago