openembedded/meta-openembedded #844: wireshark is not using qt5 anymore (by default)

FederAndInk, 2024-07-18:
When trying to build wireshark with qt5 packageconfig enabled, I get this error:

```
| CMake Error at CMakeLists.txt:1242 (find_package):
|   By not providing "FindQt6.cmake" in CMAKE_MODULE_PATH this project has
|   asked CMake to find a package configuration file provided by "Qt6", but
|   CMake did not find one.
|
|   Could not find a package conf-- Configuring incomplete, errors occurred!
| iguration file provided by "Qt6" with any of
|   the following names:
|
|     Qt6Config.cmake
|     qt6-config.cmake
|
|   Add the installation prefix of "Qt6" to CMAKE_PREFIX_PATH or set "Qt6_DIR"
|   to a directory containing one of the above files.  If "Qt6" provides a
|   separate development package or SDK, be sure it has been installed.
|
|
| WARNING: exit code 1 from a shell command.
```

Wireshark is now using qt6 by default, there appears to be an option to build for qt5 too. Maybe a modification to the qt5 packageconfig is needed? And qt6 could be added too?

See their cmake options: https://gitlab.com/wireshark/wireshark/-/blob/v4.2.6/CMakeOptions.txt?ref_type=tags

FederAndInk, 2024-07-18:
using this package config works (using `-DUSE_qt6=OFF`):

```
PACKAGECONFIG[qt5] = "-DENABLE_QT5=ON -DBUILD_wireshark=ON -DUSE_qt6=OFF, -DENABLE_QT5=OFF -DBUILD_wireshark=OFF, qttools-native qtmultimedia qtsvg"
```

I think adding a qt6 option while keeping the qt5 option would be great.

gmange, 2024-12-05:
Issue fixed on master :
- wireshark: fix Qt5/6 configuration
> https://git.openembedded.org/meta-openembedded/commit/?id=cc83b9ac17276c6ebb590a7e84e86239b9836e0f
- wireshark: use inherit_defer for conditional cases
> https://git.openembedded.org/meta-openembedded/commit/?id=97318e27ebf9f528acf77fb0759c332bd83adb4b

OldManYellsAtCloud, 2025-05-14:
@sergeevabc You mean to share the recipe? There are links for the fixes above, you can build it for your project using the latest version of the project.
Or you mean a Windows 7 build? This repository is about building your custom Linux system... possibly you can use the info from the recipe to build it yourself with qt5 on windows, but in general this project doesn't do anything with Windows.

closed 2026-10-26 by kraj