openembedded/meta-openembedded #860: polkit: including system-auth in pam file instead of common-*

FederAndInk, 2024-08-12:
Polkit pam file is trying to include non existant system-auth, if I understand correctly, OE is using common-* pam files instead.

This was working fine on kirkstone with a patch, but it was removed with the 0.119 recipe in https://git.openembedded.org/meta-openembedded/commit/meta-oe/recipes-extended/polkit?id=02e20ddb169fea0c387f70e1ba70ef676082edc0

The removed patch: https://git.openembedded.org/meta-openembedded/tree/meta-oe/recipes-extended/polkit/polkit/polkit-1_pam.patch?id=1f2c2ebcbcf5944225462f530e0d56b20f1b4630

Should it be re-included in the polkit recipe, or was it removed/not put on purpose?

kraj, 2024-08-12:
hmm it might be a fallout from removal, we should forward port the said patch to current version of polkit

ycongal-smile, 2024-09-02:
I'll look into this

ycongal-smile, 2024-09-05:
Two alternative proposals (following Peter Kjellerstedt review): 

- [meta-oe][PATCH 1/2] polkit: Switch PAM files to common-* => https://lists.openembedded.org/g/openembedded-devel/message/112093
- [meta-oe][PATCH v2 1/2] polkit: Switch PAM files to common-* => https://lists.openembedded.org/g/openembedded-devel/message/112099

closed 2024-09-07 by halstead